emoncms/emoncms #288: Bufferedwrite fails to update some feeds

inggaro, 2015-02-10:
Hi all,
I'm try to move from the full version of emoncms to the bufferedwrite branch (8.4.0 - buf 02).

It works but not all the feeds are updated; they work correctly if I switch back to the full branch. It's so strange because the inputs are correctly updated, even if I create a new feed for them the new feed is inactive.

[image]

Any ideas?

Thanks a lot

Garo

TrystanLea, 2015-02-17:
Hello inggaro, what does your process list look like, could the issue be down to differences in the available process list functions in the buffered write branch perhaps?

inggaro, 2015-02-17:
Thanks a lot for the answer! Are you just asking for my active process list? I mean something like "ps aux | less"?

Thanks again

Jerome-github, 2015-02-17:
I think Trystan meant the list of available data process. The process you use to turn Inputs into Feeds.
The dropdown menu when you configure your Feeds.

inggaro, 2015-02-17:
Sorry I didn't undestand :)

Ok so this is a picture of my inputs with processes.
[image]

They're all the same: all "log to feed" and args are somethink like "null: t_ext". But, for example, T_ext works and T_1 don't.

Even if I create a new process, it doesn't work.

Thanks again!

Jerome-github, 2015-02-17:
I'm not sure, but maybe he was asking for a screenshot of the dropdown menu with the available processing functions.

If you create a new process, you get to choose processing functions (sum, multiply, log to feed, etc.) from this dropdown.

I may be wrong.

inggaro, 2015-02-17:
Ok thank you. Here we are:
[image]
[image]

I can create new or add to the existing feed, in PHPTIMESERIES or PHPFINA formats.

TrystanLea, 2015-02-17:
Thanks Inggaro, and for helping Jerome. Can you try deleting the histogram process, that's probably it as histograms are not available on the buffered write version

inggaro, 2015-02-17:
Tried, but still not working. However, I think the problem it's near, because I just find out that using buffered version I can't configure "power" input processes (now only log and kwhd). I can edit when using non buffered version!

TrystanLea, 2015-02-17:
Can you delete the power input and reconfigure it with the log input process?

inggaro, 2015-02-17:
Wonderful!! It's working!!! Thanks a lot.

It seems I cannot recreate kwhd process however... is it changed/removed?

inggaro, 2015-02-17:
Sorry I think I find out... Is it "Wh Accumulator"?

TrystanLea, 2015-02-17:
great, yes that's it!

closed 2015-02-17 by TrystanLea

inggaro, 2015-02-17:
Thanks again!!