emoncms/emoncms #297: Request to allow user names of 3 chars - i.e. ben

sumnerboy12, 2015-02-18:
Currently I can't have a username of ben since the min user name length is 4 chars.

TrystanLea, 2015-02-24:
Great, yes I will make this change!

Jerome-github, 2015-02-24:
Hi.

I didn't check, but I was wondering. Is there a test preventing the creation of a user named "myelectric", or "node", or "vis", or such ?

What about future such reserved names, if they were added after the user creation ?

Couldn't there be URL clash between a user's dashboard and a page with the same name ?

TrystanLea, 2015-02-24:
Hello Jerome! there is no check at the moment but its the modules that are given priority. If a user creates an account with a module name they will at present be unable to view dashboards on using their username as it will go to the actual module controller first and ignore the dashboard public dashboard code.

sumnerboy12, 2015-02-24:
Thanks Trystan!

chaveiro, 2015-11-27:
Implemented on 9.2+

closed 2015-11-27 by chaveiro