emoncms/emoncms #298: Embedding visualization doesn't work unless you are actually logged in to emoncms

mharizanov, 2015-02-24:
Hi guys,

Embedding visualization doesn't work unless you are actually logged in to emoncms,

navigating to
http://emoncms.org/dashboard/view/heating?embed=1&apikey=**my_RO_APIKEY**

results in empty screen

TrystanLea, 2015-02-24:
Thanks Mharizanov, not sure why thats stopped working but can confirm Im seeing the same. I will look into it

MeroFilosofo, 2015-04-12:
Same problem here.
Empty visualization.

Thanks for your suport.

jostor, 2015-11-12:
I am also having the same problem. Any news on when it will be fixed?

chaveiro, 2015-12-30:
Feeds need to be public or you got a warning of invalid authentication. (as of 9.x)

closed 2015-12-30 by chaveiro

chaveiro, 2015-12-30:
A quick note, dashboard needs to be public also, a alias should be created and url must be acessed by username:
http://emoncms.org/username/alias?embed=1