emoncms/emoncms #300: Locally testing EmonCms

psychok7, 2015-03-09:
So i succefully installed EmonCMS locally on my machine following your tutorial. I do not yet have the "gadgets" to fetch real data, so i tried fiddling with the mysql DB and adding some values manually. It seems to me i was able to add `inputs` and `feeds` this way, but when i try to go to the `my electric` nothing shows up and i have tried all combinations there.
I probably missed something but i cant seem to figure this out. Here are some screenshots.

[image]

[image]

[image]

Any ideas what im doing wrong?

Paul-Reed, 2015-12-10:
This is a hypothetical question as emoncms is not intended to be populated in this manner.  
If when you have the hardware to gather real data you still have problems, please resubmit.

closed 2015-12-10 by Paul-Reed

Jerome-github, 2015-12-10:
To use EmonCMS without the "gadgets", you may use the input or feed API rather than play directly with the database. See API documentation in EmonCMS itself.

(Beware that API doc might not be totally up-to-date and I don't think feed API is really supported.)