emoncms/emoncms #301: error message after timeout

pb66, 2015-03-10:
More of an observation than anything as it causes no operational issues.
When logged into emoncms, If any page is left open beyond the timeout and then any of the menu bar options are clicked (user unaware timeout has occurred) emoncms will go to a blank page with one "log in" option on the menu bar. 

That is unless you click "admin" that will produce this error message

Notice: Undefined variable: result in /var/www/emoncms/Modules/admin/admin_controller.php on line 80

the "log in" button is still there and it operates the same. Maybe this module is different or the error just isn't caught? Only happens on admin any others are a blank white screen.

Paul

pb66, 2015-03-10:
PS is it possible for emoncms to bypass the blank white screen and extra click to go direct to the log in box ? Very minor point but "nicer" :-)

chaveiro, 2015-11-27:
Should be fixed after 9.0+.

closed 2015-11-27 by chaveiro