emoncms/emoncms #302: 'My Solar' not working with https

Paul-Reed, 2015-03-13:
Just tried revised 'newgetdatamethod app' but the page is being blocked because a endpoint is directed back via http instead of https. The first version of 'My Solar' worked OK.  The screenshot below probably makes more sense than me!!
[image]

chaveiro, 2015-03-21:
That is a potential security issue being blocked by the browser.
Implementation must be revised.

Paul-Reed, 2015-03-22:
A clean git emoncms install tonight revealed the same issue exists with both v1 & v2 of 'my solar', and both versions are browser blocked over https because of security concerns.  
All of the errors seem to point to jquery-1.9.0.min.js which is a depreciated script.  
  
Paul

TrystanLea, 2015-03-23:
Hello Paul, are you connecting initially with https as the error suggests? does the error disappear if you connect via http only

Paul-Reed, 2015-03-23:
There is no http access due to a rewrite rule in apache, but if I comment out the rule (to enable http access again), and load a fresh install of emoncms via http, which has been moved to the 'newgetdatamethod' branch, and add the new 'app' folder, the My Solar page still does not load - see error screenshot attached.  
All other emoncms pages can be accessed OK and appear to be working fine, but any of the 'App' pages are blank, except for the header, menu and footer.  (I've emailed you a link)
[image]

Paul-Reed, 2015-03-27:
Issue resolved, Thanks.

closed 2015-03-27 by Paul-Reed