emoncms/emoncms #311: v8.5 bugs

Paul-Reed, 2015-03-28:
A few bugs noticed following a clean git install of v8.4 switched to v8.5 branch.
1. **Mobile drop down menu**  when users are switching from any of the main menu items (Inputs, Feeds, Dashboard etc) to an 'app' sub menu, the menu works fine, however if you try to switch from one 'app' submenu to another submenu item ('My Electric' to 'My Solar'), then the main dropdown menu does not collapse. Bug apparent on Android phone, iPhone & iPad.
2. **My Electric** 'USE TODAY' constantly displays 0.0 kWh and does not increment throughout the day. The graph displays the correct data columns for the preceding days. The source feed is PHPFINA which increments as expected.
3. The **branch switch** from v8.4 to v8.5 did not go as expected, as I ran the 'Admin>Update Database' which appeared successful, but when I ran it again, it again invited me to update the db. This repeated over and over. I then checked 'Account>Timezone' which was a blank text box with no data within. I checked my emoncms/mysql user permissions which were already set to GRANT ALL. Eventually resolved by logging into mysql (using emoncms user login) and running the db update manually;
    mysql> ALTER TABLE `users` MODIFY `timezone` VARCHAR( 64 ) DEFAULT 'UTC';  
Paul

chaveiro, 2015-03-28:
No 3 fixed here: https://github.com/chaveiro/emoncms/commit/cd73f08504a718da14abc92d6951d96b7653d9aa

TrystanLea, 2015-03-29:
Thanks Chaveiro, I've copied your fix over to the v8.5 branch too for bug 3. Thanks Paul!

Paul-Reed, 2015-03-29:
A quick response Chaveiro, thanks.   
I've added a few more...   
1. ***Mobile drop down menu*** when users are switching from any of the main menu items (Inputs, Feeds, Dashboard etc) to an 'app' sub menu, the menu works fine, however if you try to switch from one 'app' submenu to another submenu item ('My Electric' to 'My Solar'), then the main dropdown menu does not collapse. Bug apparent on Android phone, iPhone & iPad.
2. ***My Electric 'USE TODAY'*** constantly displays 0.0 kWh and does not increment throughout the day. The graph displays the correct data columns for the preceding days. The source feed is PHPFINA which increments as expected.
3. When viewing feed data via mobile devices a ***'Undefined offset error'*** message appears at the top of the page - [see screenshot](http://openenergymonitor.org/emon/node/10381#comment-29047)
4. Again, only on mobiles... the ***'My Solar' app does not format well*** and the graph shrinks to the left instead of full width - [see screenshot](http://openenergymonitor.org/emon/node/10381#comment-29000)   
Paul

TrystanLea, 2015-03-31:
Hello Paul, mysolar resizes much better now, although still not perfect.

Paul-Reed, 2015-05-11:
Are there any updates to the bugs listed 29th March above.

Thanks

Paul

chaveiro, 2015-11-27:
Are this issues fixed on 9.2 ?

Paul-Reed, 2015-12-10:
Now fixed

closed 2015-12-10 by Paul-Reed