emoncms/emoncms #314: Daily feeds not displaying any data prior to DST change

Paul-Reed, 2015-04-09:
All daily feed graphs are showing no data prior to 30th March (DST was 29th March).   
Realtime data is however OK, and appeared to handle the DST OK.   
[image]

chaveiro, 2015-04-10:
Hi, give a public link to see that chart live with the issue.

Paul-Reed, 2015-04-10:
The link is https://
<https://www.digitalnut.co.uk/emoncms/vis/auto?feedid=30>
www.digitalnut.co.uk
<https://www.digitalnut.co.uk/emoncms/vis/auto?feedid=30>/emoncms/vis/auto?
<https://www.digitalnut.co.uk/emoncms/vis/auto?feedid=30>feedid
<https://www.digitalnut.co.uk/emoncms/vis/auto?feedid=30>=30
<https://www.digitalnut.co.uk/emoncms/vis/auto?feedid=30> or
https://www.digitalnut.co.uk/emoncms/vis/auto?feedid=16

Hi, give a public link to see that chart live with the issue.

chaveiro, 2015-04-10:
Hi, you have missing data from 4Jan to 29 Mai. Service returns null for thees days. I believe some file may got corrupted in 4 jan and started working on dst change day due to software upgrade.
What engine type you have? Does the engine has a separate file for each resolution? If so the day resolution file may have got corrupted on that ranges.

Paul-Reed, 2015-04-10:
Thanks Chaveiro. I have 6 'Daily' feeds (15. 16, 28, 30, 31 & 32) all of which are phptimeseries, and all were recording fine up to 4 Jan (although shows 3rd Jan when hovering over barchart!), then stopped, and started working again 30th March.  How likely is it that all 6 feeds became corrupted at the same time?   
The phptimeseries files are just one file per feed, and are stored in a non human readable format, so I cannot examine it's contents.   
It's no major issue for me to lose this data, but would like to know what had caused it.   
Paul

TrystanLea, 2015-04-10:
Hello Paul, Chaveiro

I think this could be due to the change in the get data request method, there may not be any data loss, I think its because the request tries to find a value within a small window of the request time and the datapoint is 1h away from the request time. Il could also explain why it only became apparent after the time shift.

Trystan

Paul-Reed, 2015-04-11:
Trystan, I think that you are correct, because if I switch from v8.5 to Master branch, all of the data going back to last September is there and intact, except for one day - 29th March (DST Changeover) which returns null.   
I've now returned back to v8.5, and as before, all data between 4th Jan and 29th March is again not shown.   
   
Paul

chaveiro, 2015-04-13:
Ok, i see the problem.
With the timezone fix the daily date is saved as local midnight converted to utc on the engine (and is correct), before it was just local midnight at server timezone. What it means is that previous data may be offset if server was not in UTC. Can you confirm your server timezone?
What i dont understand with your problem is what happened at 4-Jan and why previous data is ok also. Any idea?
I use mysql engine and i noticed that after the fix all previous date required an offset fix because of the above. For the other engines it's expected to require the date offset fix too, but i cant help much as i dont have data for any of those engines.

Paul-Reed, 2015-04-13:
My Timezone is;   
Current default time zone: 'Europe/London'
Local time is now:      Mon Apr 13 17:31:31 BST 2015.
Universal Time is now:  Mon Apr 13 16:31:31 UTC 2015.   
I have no idea what changed 4-Jan. I usually pull git changes pretty quickly, but can't see anything around that time relevant.  
Not sure how phptimestore feeds can be offset fixed??    
As I have said previously, I don't mind losing the data and starting afresh, but if it's something buggy in v8.5, I can imagine a lot of users being unhappy if they lost their historical data.

Paul-Reed, 2015-04-13:
I've just decoded one of the phptimeseries feeds using a Python script written by Trystan, and imported the data into Excel, converting the unix timestamps to GMT, I got the following daily time stamps;   
Up to 26/10/2014 = 22:00:00hrs
26/10/2014 - 3/1/2015 = 23:00:00hrs
(BST changeover to GMT
4/1/2015 - 28/3/2015 = 00:00:00hrs
(no idea what changed!)
29/3/2015 - present = 23:00:00hrs
(GMT changeover to BST)

The current timestamp (23:00:00hrs) I presume is correct as we are now GMT+1, but thought that the data up to 26/10/2014 should have been 00:00:00hrs, and have no explanation as to the period 4/1/2015 to 28/3/2015.

Paul

chaveiro, 2015-04-14:
Ok, then it's something wrong with the new 8.5 method of reading dates from the phptimeseries engine timestamp.
I'm sure Trystan will figure it out quickly.

TrystanLea, 2015-04-15:
Trying to figure out whats not working here, I have a couple of daily phptimeseries feeds on emoncms.org that are working fine, do we have any examples of other installs or anyone who has mentioned this relating to emoncms.org?

pb66, 2015-04-15:
There may be an indirect link between this issue and https://github.com/emoncms/emoncms/issues/199. The [Daily graphs not displaying correct date](http://openenergymonitor.org/emon/node/6422) forum thread is dated 4th of Jan and may of led to something being changed in an attempt to fix that issue that got undone with the recent updates.

Frankster-NL, 2015-05-05:
I have the exact same issue (as reported on the forum at: http://openenergymonitor.org/emon/node/10508) The graphs below will show you the difference. I basically did not touch the data folder, but merely installed the 8.5 verion (both the 'normal' and the 'extended')

[image]

[image]

Frankster-NL, 2015-10-14:
@paul: where did you find the script to decode phptimeseries? As this issue is still not solved I would like to check if my old data is still intact now I have moved to version 9 (which has the same issue)

Paul-Reed, 2015-10-14:
@Frankster-NL  
I thought that it was a script within https://github.com/emoncms/usefulscripts but checking tonight I can't locate it, sorry.

Paul-Reed, 2015-10-15:
@Frankster-NL 
The script is at https://github.com/emoncms/usefulscripts/blob/master/enginereaders/phptimeseries.php

Paul

Frankster-NL, 2015-11-04:
As indicaties in the forum: I van conform that, after converting the deed with the script, that all 'older' data is corrupted. Time values appear OK, but the values are all zeros. Will test the latest RC again and report back.

closed 2015-11-27 by chaveiro