emoncms/emoncms #330: I have same error

kumarmohan7, 2015-05-20:
Can't connect to database, please verify credentials/configuration in settings.php
Error message: Can't connect to MySQL server on 'smk-iot.16mb.com' (111 "Connection refused")

chaveiro, 2015-05-20:
If user and pass are ok, you may need to allow access by ip address in mysql :
GRANT ALL ON yourdatabasename.* TO user@'%' IDENTIFIED BY 'yourPassword';
Ask for support on your hosting company if not allowed to execute the command.

kumarmohan7, 2015-05-21:
Than you for reply

Please give mysql database files and steps

kumarmohan7, 2015-05-21:
Fatal error: Call to a member function fetch_array() on a non-object in /home/u645649278/public_html/emoncms-8.5/Modules/user/user_model.php on line 507

Please give mysql database files and settings

closed 2015-11-27 by chaveiro