emoncms/emoncms #337: Post multiple values in JSON format

vosmont, 2015-06-05:
Hello,
is it possible to send multiples values in JSON format, with a timestamp ?

Perhaps something like that :
http://emoncms.org/input/post.json?json={{power:200,offset:-10},{temperature:25,offset: -5}}&time=1433504598

vosmont, 2015-06-05:
It seems that
http://emoncms.org/input/post.json?json={power:200,temperature:25}&time=1433506452
at least should work, but when I set a time different from current time, the values are not stored.

vosmont, 2015-06-05:
In fact, the value is stored only if it is after the last recorded value.
You cannot thus insert values post factum.

So my need is to be able to set an offset on the sent values.

arthurlutz, 2015-08-26:
I think I've bumped into the same bug as you @vosmont see #368 and #371, have you found a fix ?

PR #645 (2017-02-28): Adds descriptions for named keys using bulk insert
Closes #337 together with #615.
closed 2017-02-28 by TrystanLea