emoncms/emoncms #346: Updating input via api creates a new input

rechena, 2015-06-19:
Seems that on version 8.5 when I call the api via: 
http://bla.com/input/post.json?json={power:200}&apikey=myapi it creates a new input line every time instead of updating the existing one.

arjenhiemstra, 2015-06-22:
I had the same thing with one of my users, proper formatting of the JSON string solved it:
{power:200} is not a valid JSON. Use {"power":200} or {"power":"200"} instead. In the latter example 200 is also indentified as string, this is not needed but supported by emoncms (emoncms converts it to a number).

rechena, 2015-06-22:
Hi, thanks for the reply, just tested with input/post.json?json={"realPower":"168.70","apparentPower":"268.82"}&apikey= and the same happens, still no joy. :/

arjenhiemstra, 2015-06-22:
I stand corrected. Just tested it with my own server, same behaviour.
Try adding node=[number], that stopped recreating the inputs for me.
/input/post.json?node=0&json={"realPower":"168.70","apparentPower":"268.82"}&apikey=

It also seems it does not matter whether you use quotes or not, but still, proper JSON formatting can't hurt.

rechena, 2015-06-22:
Nice, it did worked, but I had to rollback again since all my charts/history disappeared, but that is out of this scope. Now on this subject then the API docs should be changed since they don't state the node=n option.

321solar, 2015-07-10:
Hey gents, I too have been messing around with the newset emon version. I only had to add the node=0 and my inputs stopped multiplying each time data was added. http://127.0.0.7/emoncms/input/post.json?node=0&json=. Thanks for the great info it saved me a bunch of time.

arthurlutz, 2015-08-15:
Got hit by this bug too. Very confusing. Thanks for the tip on node=0 which works. This should work out of the box or be explained in the API documentation.

closed 2015-11-27 by chaveiro