emoncms/emoncms #347: EmonLogger doesn't create a file if it doesn't exist

cagabi, 2015-07-01:
Hi there,

We are writing a module that logs info into a file using the newest version of EmonLogger. What we have done so far is to manually create the log file in the module folder. So the log file comes by default when you install the module.

The problem: we want to make the log file accesible in the User Interface. Having only one log file means that we will log info from different users in the same file. To display the contents of the file to a specific user would reveal data about other users activity and that is wrong.

So what we want to do is to use one log file per user, so in the beginning of the module we will have:
```
include "Modules/log/EmonLogger.php";
$log = new EmonLogger();
$log->set_logfile(__DIR__ . '/rules_' . $session['userid].'log');
```
EmonLogger uses `fopen($file_name, 'a')` which will create the file if it doesn't exist. But when trying it, the file is not created. Having a close look to EmonLogger i found out why. The following code is for the `info($message)` method but is the same for the `warn($message)` method 
```
public function info ($message){
   if (!is_writable($this->logfile))
         return;
   if ($fh = fopen($this->logfile,"a")) {
         fwrite($fh,date("Y-n-j H:i:s", time())." $this->topic INFO ".$message."\n");
         fclose($fh);
    }
}
```
So our new log files are not created because the first *if* statement is checking if the file is writable, obviously as the file doesn't exist yet, it is not writable then we return and do nothing.

The proposed solution:

```
public function set_logfile($logfile) {
	if (!file_exists($logfile)) {
		$fh = fopen($logfile, "a");
		if ($fh) {
			fclose($fh);
            $this->logfile = $logfile;
            return true;
		} else
			return false;
	} else {
		$this->logfile = $logfile;
        return true;
	}
}
```
What  we are doing is checking if the file exist, if not we try to create it (opening it). On success we set the logfile to point to the new file, if the file was not created (for example if there are not writing permissions for the directory) we return false. If the file exist we simply set the log file. I have tried it and works ok.

Some more thoughts
- If the writting permissions of the directory are limited, we will get a warning, the file will not be created and EmonLogger will not set the new file to use.
- To avoid this, the best practice is to have the log file already in the module and always use that one instead of creating new ones.

   Carlos

chaveiro, 2015-11-27:
If problem is writing permission there is nothing emoncms can do.
Warning and fix instructions are given in administration page.

chaveiro, 2015-11-27:
Ok after reading again this issue you have a new requirement, maybe you can re-enable logenabled on that set_logfile() function after it has successful created/has write permission to the file you want.

closed 2015-11-27 by chaveiro