emoncms/emoncms #352: input/bulk.json errors with bad or missing apikey

sheffieldnick, 2015-07-12:
This is on emoncms.org (v8.3.5):

1. Call bulk input with an incorrect apikey string, and instead of trapping the error, the API returns
"Notice: Undefined index: userid in /var/www/emoncms/index.php on line 131"

2. Call bulk input with no apikey argument at all, and you get a 200 OK response but 0-length content. Should be a human-readable error message about the missing argument.

sheffieldnick, 2015-07-12:
3. Error messages not being returned if invalid &data= argument. I'm always getting "ok" response no matter what rubbish I'm passing to the API !
a) Missing &data= argument entirely
b) Invalid JSON in &data=
c) Not enough data values (less than 3) in &data=
d) Last item in &data= array is empty

luizlaf, 2015-08-02:
Fixed on newer versions

closed 2015-11-27 by chaveiro