emoncms/emoncms #358: remember me is not working

dazzag24, 2015-07-23:
"Remember me" isn't working. I use the emoncms.org site and access it from several devices.

Windows PC - Google Chrome 41.0.2236.0 dev-m
Linux - Google Chrome 39
Android Phone - Chome 39.0.2171.93

All of them refuse the remember the login details which means that I am repeatedly having to enter my details. This makes the use of the dashboard view pretty pointless as I can't simply load it somewhere and let it run indefinitely, as I need to keep logging in again and again.

Thanks

yekorzh, 2015-07-23:
I've noticed this issue on my standalone emoncms 8.5.2 as well. After some time I have to re-login even if I selected "Remember me" option.  

Chrome on Windows
Chrome on Android

Thanks

chaveiro, 2015-11-27:
should be ok on 9.0+

closed 2015-11-27 by chaveiro

arthurlutz, 2016-03-07:
am having this on a v9.2

Paul-Reed, 2016-03-07:
Seems to work OK here on laptop, phone & tablet. Are you clearing cookies when closing your browser? can you try a different device.