emoncms/emoncms #359: bargraph label should use 24 hour time format

yekorzh, 2015-07-23:
Hi,

I'm using bargraph to display hourly power consumption. When I point mouse cursor over the bar, it displays text showing total power during that period, date and time. It looks like time is in 12-hour format and doesn't show am or pm, making it very confusing. I suggest changing the time format to 24 hour to match values on x-axis.

[image]

pb66, 2015-07-23:
Can you double check that? I have seen a issue on the forum that may suggest the "chart" time might be 12hrs out.

yekorzh, 2015-07-23:
Here is another highlighted bar from the same chart on the same date. Notice that the time is the same. 
[image]

Regarding 12 hour issue, I think in my case all the data corresponds to correct time on the x-axis, what I see however is that when I open the dashboard chart doest show most recent 12 hours of data , but when I  click ">"  graph scrolls to the right an data appears.

PR #457 (2016-01-23): Set hover text to 24h format
Fixes #359
closed 2016-01-23 by glynhudson