emoncms/emoncms #365: Feed update giving error when event is assigned to the feed

Rishabh-K, 2015-08-06:
Hello I am new to emoncms.
I have setup emoncms on xampp windows.
Created a feed named "feed1" 
I created a event on this feed
[image]

I am using url (to pass data to thiis feed):
http://localhost/emoncms/feed/update.json?id=6&apikey=xxxxxxxxxxxxxxxxx&value=70.0

But the window is showing an error:
Fatal error: Call to a member function get() on null in C:\xampp\htdocs\emoncms\Modules\event\event_model.php on line 239

This error is showing only when an event is associated to this feed.

This error vanishes when no event is associated with this feed.

Second thing the event works fine i.e. conversion of false to true when condition is valid but the transformation of false to true is visible after page is manually refreshed.

[image]

Third thing is the event fires but the action assigned to it i.e. setFeed does not fires.

elyobelyob, 2015-08-06:
There is currently an issue with event. Someone has changed the db schema
and it broke the system.

Hopefully someone can take ownership of fixing, I've not got time. I did
post some info yesterday to another discussion r.e. this.

Paul-Reed, 2015-12-10:
A number of event module incompatibilities has occurred as emoncms has evolved to the current version. A replacement event module is currently under development, and will hopefully be published shortly.

closed 2015-12-10 by Paul-Reed