emoncms/emoncms #366: Timestoredaily shows wrong value for last day in series

jesjimher, 2015-08-10:
I've been using 9.0 preview for a few days, and is great, but I've found a little problem with the new, beta, timestoredaily visualization. While it works mostly without problems, I've found that when you use it to calculate kWh per day from a W feed, the last graph value (which by default shows current day kWh) is wrong, and it's much bigger than it should be. Previous "power to kWhd" method shows the right value (which I validated by manually calculating it in a spreedsheet from a CSV export).

Looking at the code, it seems that timestoredaily code (timestoredaily.php, draw() function at line 131) assumes that all average values returned by get_feed_data correspond to full days, so they're multiplied by 0.024 (24/1000) to obtain the final kWh value for each day. The problem comes when the last day in the interval has only partial data, and it's still getting multiplied by 24 hours. The correct behaviour would be multiplying this last value not by 24, but by the maximum time we have data for. For example, if last W value is at 09 AM, it should be multiplied by 0.009.

I would provide a patch/pull request with the fix, but unfortunately I'm not familiar enough with emoncms API to be sure I'm not breaking other people timestoredaily use cases/dashboards.

Thanks for your attention and keep up the good work with emoncms. 9.0 looks better than ever!

Paul-Reed, 2015-12-12:
Is this still an issue in v9.2?

jesjimher, 2016-01-13:
Yes it is. Right now, at 15:31h, with approximately 17 kWh spent today, timestoredaily (wrongly) shows 25 kWh.

PR #445 (2016-01-16): Fixes timestoredaily showing wrong value for present day
Fixes #366
closed 2016-01-16 by chaveiro