emoncms/emoncms #368: Insert missing values between dates already recorded in feeds

arthurlutz, 2015-08-24:
My use case is the following : 

I have a device that posts values to emoncms using the HTTP post json values. From time to time the device will be offline and when it does it records the data by storing on disk the json and the timestamp (`date +%s`), the idea is that when it comes back online it starts posting the live data again and I then process the old data. 

It appears that the old data is not inserted into the database. 

Is this by design of the feeds ? If so it should probably be mentioned in the Input API help. 

If it's not a bug, I need to find a way to push all old values before reactivating the live http posts ? If so, do you know a project that does that since right now my design is pretty simple (test the eth0 status, if up do http post, if down echo http post to a file) ?

mpruessmeier, 2015-08-25:
Yes, there is a Check in the Processing that lolder data are not processed. I changed that for my installation. And than it would work for you.

arthurlutz, 2015-08-25:
@mpruessmeier thanks for your reply. Glad to know it is possible. Did you do a code modification or there is a configuration in the application ? (can't find it)

mpruessmeier, 2015-08-25:
Hi @arthurlutz,

in process_model.php ther is often a Phrase checking last_time. I modified it in the progeam to:

//        if ($last_time && (time()-$last_time)<7200)
//        auf ein Jahr angepasst !!!
        if ($last_time && (time()-$last_time)<(7200*12*365))
        {
Than you get an year back

arthurlutz, 2015-08-25:
@mpruessmeier I'm not sure this solves my problem. I have "Log to feed" on inputs, not "Power to kWh" which your patch seems to improve (when I get to that transformation I might be interested in this change and maybe it could configurable) 

Looking though the processes, I found update feed @time, tried that but it seems to crash on an error, which I'm adding to another issue on github.

Paul-Reed, 2015-12-17:
Arthur, have you revisited your outstanding issues in V9 yet. If so, could you please confirm if they are still valid.

closed 2017-02-28 by TrystanLea