emoncms/emoncms #369: No accents allowed in inputs / dashboard names

arthurlutz, 2015-08-24:
It would be nice to be able to add descriptions for inputs with accents (for french speaking users for example), same goes for dashboard names and various other places in emoncms.

Jerome-github, 2015-08-24:
I thought those were fixed with the move to UTF-8.

https://github.com/emoncms/emoncms/issues/84
https://github.com/emoncms/emoncms/issues/48
https://github.com/emoncms/emoncms/issues/17

If your database was created before UTF-8 became default, see https://github.com/emoncms/emoncms/issues/85 for the transition.

euqip, 2015-08-24:
I think the problem is located in the regex filter used to clean data before storage.
example out of user_model.php
        $flds .= '  location = "'.preg_replace(REGEX_STRING_ACCENT,'',$data['location']).'"';
        $flds .= ', language = "'.preg_replace(REGEX_STRING,'',$data['language']).'"';

where 

define('REGEX_STRING','/[^\w\s-]/');
define('REGEX_STRING_ACCENT','/[^\s\p{L}0-9]-\'/u');

these are to be defined once at the strart  (and correctable or adjustable once), to be usable anywhere.
 
Effectively the database needs also to accept utf-8 characters because accents are not allowed in ASCII.

Hope it will help

Paul-Reed, 2015-12-12:
Arthur, you've currently 5 open issues, and I'm not alone in thinking that at least some of the issues may have already been resolved through a series of bug fixes as emoncms has evolved into the current stable version - v9.  
Could I ask that you consider updating your installation to the current version, otherwise we may fall into the trap of duplicating the development work which has already taken place.  
We would be grateful if you could update the open issues with the results please.

arthurlutz, 2015-12-14:
@Paul-Reed hi, I've found some time to upgrade to 9.2. I've successfully renamed my dashboard to something with accents. Closing this ticket. 

Will try to find some time this afternoon to go through my other ticket. Thanks.

closed 2015-12-14 by arthurlutz

Paul-Reed, 2015-12-14:
Much appreciated Arthur.