emoncms/emoncms #386: Theme/Basic Issue

jldeen, 2015-09-06:
Hi there,

Just installed on Ubuntu 14.04 - double checked everything, but I seem to be getting this error at the top of every page: Notice: Undefined index: order in /var/www/emoncms/Theme/basic/menu_view.php on line 97

Thoughts?

JSidrach, 2015-09-06:
Hi, could you tell me which modules do you have installed? (`ls /var/www/emoncms/Modules/`)

krichot, 2015-09-07:
Same here:
Modules loaded
admin, energy, feed, mqtt, openbem, process, report, time, vis, dashboard, event, input, notify, packetgen, raspberrypi, schedule, user

JSidrach, 2015-09-07:
Hi. I just checked every module that you mentioned (except time, I have not found the repository for that one). The following modules are the cause of the issue:
* wifi
* notify

I am not familiar with these modules, but the fix is an easy one. Go to `Modules/wifi/wifi_menu.php` and `Modules/notify/notify_menu.php` and add to the $menu_right  'order' => 20 (I made up that number, but the lower it is the more preference it has showing up in the top bar).

Something like this (one line for each file):

`$menu_right[] = array('name'=> dgettext($domain, "Wifi"), 'path'=>"wifi" , 'session'=>"write", 'order' => 20);`

`$menu_dropdown[] = array('name'=> dgettext($domain, _("Notify")), 'path'=>"notify" , 'session'=>"write", 'order' => 20);`

I am not a dev of these modules but it may be useful if any of you could create a pull request with these changes too :smile:

jldeen, 2015-09-07:
Hi. I'm going to try this right now. I don't have the wifi module loaded, but I do have notify loaded. I only loaded the modules listed in the Install documentation:

 admin
 dashboard
 energy
 event
 feed
 input
 mqtt
 notify
 openbem
 packetgen
 process
 raspberrypi
 report
 schedule
 time
 user
 vis

Thanks!

JSidrach, 2015-09-07:
Ok, let me know if the error persist

krichot, 2015-09-08:
I'd like to confirm - problem solved, I had to edit notify module only.

JSidrach, 2015-09-08:
Great! Consider opening a Pull Request on the notify module to fix the issue for other users too :smile:

Nargousias, 2015-10-02:
Just performed a new install on a clean system.

Getting the two lines at the top of all pages with two instances of:

Notice: Undefined index: order in /var/www/emoncms/Theme/basic/menu_view.php on line 97

I have the following modules running:

admin
dashboard
energy
event
feed
input
mqtt
notify
openbem
packetgen
process
raspberrypi
report
schedule
time
user
vis
pi

I have performed the adding of the order statement in the notify_menu.php.

I have checked all of the other _menu.php files and they have the order statement.

Any additional recommendations to try?

JSidrach, 2015-10-02:
Hi @Nargousias 
I have read the code again and the only possible reason you are getting the notice is because there is still some _menu.php with no 'order'. Could you double check? Could you paste the code you modified? I will have a look at it

Nargousias, 2015-10-02:
<?php

  $domain = "messages";
  bindtextdomain($domain, "Modules/user/locale");
  bind_textdomain_codeset($domain, 'UTF-8');

  $menu_dropdown[] = array('name'=> dgettext($domain, _("Notify")), 'path'=>"notify" , 'session'=>"write" , 'order' >= 20 );

?>

JSidrach, 2015-10-02:
I think you just have a small typo, it should be `'order' => 20`, not `'order' >= 20`
Hopefully this solves your issue, just tell me if it is ok now or not :smiley:

Nargousias, 2015-10-02:
I do have two different dashboard_menu files in diffenent locations

the one in /var/www/emoncms/Modules/dashboard has this line:
    $menu_dropdown_config[] = array('name'=> dgettext($domain, "Dashboards"), 'icon'=>'icon-th-large', 'path'=>"dashboard/list" , 'session'=>"write", 'order'=>30 );

The one in /var/www/emoncms/Modules/dashboard/Views does not

JSidrach, 2015-10-02:
Have you tried what I just wrote before?

closed 2015-11-27 by chaveiro