emoncms/emoncms #403: Only one value for timeseries feed

arthurlutz, 2015-11-09:
Hi, 

I'm having a problem when trying to graph a value by day on emoncms 8.5.2  2015.07.20

[image]

Here is the feeds view : 

[image]

and the transforms from the input 

[image]

The "transformation" is inspired by Fred's comment that can be found here  http://hallard.me/teleinfo-emoncms/#comment-10290

```
1 -> Log to Feed – Compteur (Utile pour la phase de dessous)
2 -> Total pulse count to pulse increment – (Diff N- N-1)
3 -> Log to Feed – Elec. Wh
4 -> Accumulator – Elec. Wh Acc
5 -> X 0.001 ( vers kWh)
6 -> kWh to kWh/d (Consommation Journalière en kWh/d)
7 -> kWh to Power (Puissance instantanée)
```

I'm getting the same problem with my teleinfo electricity counter. Values seem ok but no graph for the daily kWh/d.

arthurlutz, 2015-12-20:
am getting the same problem with v9

chaveiro, 2016-02-09:
May be related to this https://github.com/emoncms/emoncms/pull/414.
Can you test with the fix?

arthurlutz, 2016-02-29:
@chaveiro #414 solves my issue. Thanks.

closed 2016-02-29 by glynhudson