emoncms/emoncms #407: Error in Accumulator Process, maybe

mpruessmeier, 2015-11-22:
I have a strange situation in process an accumulation.
Every 10 seconds i get a value of 50 to 53 Wh as input.
In processlist I multiply this by 0.001 for kWh and accumulate into a new feed.
The value in the new feed growed like whished, but now it goes over 10000 in value in the new feed.
Now the value seems to be cutted or rounded ?!? 
The input stays at 52 Wh but there are added 100Wh every 10 seconds by the accumulator.
this started when the value switched from 9999 to 10.000 ?!?

Paul-Reed, 2015-12-12:
Are you running Redis on your system?

mpruessmeier, 2016-01-13:
Yes, Redis is running with redis-cli 2.8.4
I have the same on a second feed reaching 10.000 and accumulating by round 0.02.
the value is calculated now every accumulation to 10.000 fixed.
My emoncms version is 9 RC2 | 2015.09.15

mpruessmeier, 2016-01-13:
It looks to be a problem in PHPTileSeries.php

With PHP there are packing and unpacking functions:
1.   fwrite($fh, pack("CIf",249,$time,$value));
2.   $array = unpack("x/Itime/fvalue",$d);
in my opinion the float value here could be to small and should be double

mpruessmeier, 2016-01-13:
if i do /feed/data.json i get this
so the float is big enough
[1452716100000,10000.01953125],[1452717000000,10000.01953125],[1452717900000,10000.01953125],[1452718800000,10000.01953125],[1452719700000,10000.01953125],[1452720600000,10000.01953125],[1452721500000,10000.01953125],[1452722400000,10000.01953125],[1452723300000,10000],[1452724200000,10000],[1452725100000,10000],[1452726000000,10000]]

mpruessmeier, 2016-01-14:
Maybe I found the problem. 
The function get_timevalue in feed/feed_model.php gets the last value from mysql - feeds. 
Instead of getting it from the engine ?!?

mpruessmeier, 2016-01-14:
I put this to the last else statement in the get_timevalue
 $lastvalue = $this->EngineClass($engine)->lastvalue($id);
now it is working correctly. Ill try this out a while

CapnBry, 2016-02-15:
I also ran into this issue, except with PHPFina and without redis. The feeds.value field in the mysql table is the single precision float that is being rounded down. In my case it was 1257.77 trying to accumulate ~0.002 which would become 1257.77 again. I solved it by altering the mysql table to convert the float to a double.

MySQL floats appear to behave slightly differently than regular floats because 1257.772 can actually be stored in 32 bits just fine, but even if I try to edit the field in phpmyadmin, the value flips back to 1257.77 if the datatype is float.

This should be changed in Modules/feed/feed_schema.php, change the datatype of value to "double" and then run the "Update Database" function from the administration console.

Paul-Reed, 2016-02-21:
Following further discussions in the [forum](http://openenergymonitor.org/emon/node/120040) the datatype has been changed to 'double' in the master branch.
Can you please check if this is now resolved.

closed 2016-10-15 by chaveiro