emoncms/emoncms #413: Timezone issues for multi users

Paul-Reed, 2015-12-01:
[Forum post](http://openenergymonitor.org/emon/node/11720) describes php errors occurring if not all users have updated their timezone setting. Is there a way to handle this?

Paul

firefox7518, 2015-12-01:
I have the same issue which I described there: http://openenergymonitor.org/emon/node/11566

As written when using the the edit daily functionality in the visualization area the data is visible! Whats the difference between the edit daily visualization code and the ones used on the dashboard?

arjenhiemstra, 2015-12-14:
It might be an idea to add a check.
Where the query("SELECT timezone FROM users WHERE id = '$userid';") is done in code to check if the result matches an item from the array DateTimeZone::listIdentifiers(), if not default to UTC and log a warning.
I will have a look at it.

closed 2017-02-28 by TrystanLea