emoncms/emoncms #420: No process description available for process 'Wh increments to kWh/d' with id '8'.

arjenhiemstra, 2015-12-07:
No process description available for process 'Wh increments to kWh/d' with id '8'.
Add a description to Module\module_name\module_name_processlist.php in process_list() function, $list[] array at the 'desc' key.

Paul-Reed, 2015-12-12:
Thanks Argen for flagging this. As this is an open source project, if you are using, or familiar with this process it would be great if you could contribute to the project by posting a suitable description here, from which I'll update the repo.

Paul
Community Contributer

arjenhiemstra, 2015-12-13:
Paul,

You can close this issue, I registered the issue as it is being asked in the UI and to work on it later, but bardjc was faster :)
Implemented already with PR 421.

Arjen

closed 2015-12-13 by arjenhiemstra

Paul-Reed, 2015-12-13:
Thanks Argen