emoncms/emoncms #426: Check trailing slash for phptimeseries, phpfina and phpfiwa directories

Jerome-github, 2015-12-13:
Minor issue.

I just noticed that there was no trailing slash in the paths in my settings.php.

        'phpfiwa'=>array(
            'datadir'=>'/home/jl/emoncms/data/phpfiwa'
        ),
        'phpfina'=>array(
            'datadir'=>'/home/jl/emoncms/data/phpfina'
        ),
        'phptimeseries'=>array(
            'datadir'=>'/home/jl/emoncms/data/phptimeseries'
        ),

and none is added in the code, so that all the data was in the root directory:

    -rw-rw---- 1 www-data www-data 1398596 juil. 26 09:23 phpfiwa22_0.dat
    -rw-rw---- 1 www-data www-data  139864 juil. 26 09:23 phpfiwa22_1.dat
    -rw-rw---- 1 www-data www-data      45 déc.  16  2014 phptimeseriesfeed_28.MYD

rather than in separate directories (phpfiwa/, etc.). See, the file name is just concatenated to the dir path with no separating slash.

I might send a PRwith a manual fix, but there may be a more PHPish way of enforcing this. Like in Python, instead of adding '/' manually, I'd rather use the os.path.join() method.

No big deal, no data loss issue. However, fixing it could be surprising for the unaware user (his paths would be "broken"), so better sooner than later IMHO.

Paul-Reed, 2015-12-13:
What operating system are you using Jerome, this has not occurred before, to my knowledge.  
Normally [once the data dirs are created & permissions are set](https://github.com/emoncms/emoncms/blob/stable/docs/RaspberryPi/readme.md#create-data-repositories-for-emoncms-feed-engines), the data lands OK in the respective directory.

Jerome-github, 2015-12-13:
Linux.

Not sure I was clear. What I'm saying is that when the user enters

    'datadir'=>'/my_path/phpfiwa'

it should be interpreted as

    'datadir'=>'/my_path/phpfiwa/'

which is not the case as of 9.2. Or maybe I totally misunderstood the whole thing.

Paul-Reed, 2015-12-13:
No, I do accept your point and agree, as I always assume that - 'if it ends with a slash, it's a directory. If not, it's a file', but I'm surprised that your data files went to root, as I've never heard in the forum of this occurring with emoncms before.

Jerome-github, 2015-12-13:
I didn't mean the root directory '/' but the root storage directory (in my example, this is '/my_path/').

In other words, instead of

    /my_path/phpfiwa/22_0.dat

I got 

    /my_path/phpfiwa22_0.dat

Sorry about the confusion. Let me edit my first post.

Jerome-github, 2015-12-14:
From a quick search, it seems there is no PHP equivalent to Python's `os.path.join()`.

http://stackoverflow.com/questions/1091107/how-to-join-filesystem-path-strings-in-php

What a shame.

chaveiro, 2015-12-30:
I I personally prefer the way it is, maybe highlight it on the documentation will help others to not forget the ending '/'.

closed 2015-12-30 by chaveiro

Jerome-github, 2015-12-30:
Rather in the comment in settings.php than in the doc, IMHO. Or both, but the comment before the paths is a good place for this.