emoncms/emoncms #429: Enable SSL for emoncms

jpsingleton, 2015-12-18:
Please enable (or even better, force) SSL / TLS / HTTPS on emoncms, *especially* for [emoncms.org](http://emoncms.org/).

[HTTPS is not supported on emoncms.org](https://emoncms.org/). **This is bad**. Power usage data is sensitive as it reveals a lot about a home and it's occupants. Things like if a property is empty or what is occurring inside, all in real-time. Not using encryption also leaks user credentials which may have been reused on other sites.

SSL is easy and free to set up (including automatic and free domain certificates) with a service like [CloudFlare](https://www.cloudflare.com/) or [Let's Encrypt](https://letsencrypt.org/). You could also use a proxy such as [stunnel](https://www.stunnel.org/index.html) which would work for local installs too. Here's a [shell script](https://github.com/wireload/screenly-ose/blob/master/misc/enable_ssl.sh) that shows how to set it up. Certificates are cheap if required but even self-signed ones would be better than the current situation.

chaveiro, 2015-12-30:
This is no bug, please open a new discussion on the forum for this.

TrystanLea, 2015-12-30:
Agreed that this is needed on emoncms.org, its been something I've wanted to look into for some time. Thanks for highlighting and the information about CloudFlare and Let's Encrypt.

Jerome-github, 2015-12-30:
I think such proposal should not be closed but kept open and marked as "feature request", "evolution" or such to keep track of ongoing work and discussions. Until it is either "done" or "wontfix" / "rejected".

TrystanLea, 2015-12-30:
Yes lets keep it open, a switch over to using SSL as default on all systems will also need changes to emonhub and recommendations in the emoncms documentation

chaveiro, 2015-12-31:
Enabling ssl is part of the http server configuration and is not a feature associated with the development of emoncms.
Have you tried enabling ssl on your private server serving emoncms?

Paul-Reed, 2015-12-31:
I also share Chaveiro's view.
This would be better logged in a development plan and the issue closed, because it's not specific to any current issue within emoncms, nor can it be addressed by any changes to emoncms code.

Jerome-github, 2015-12-31:
You've got a point: this is not really related to EmonCMS code. Unless we'd like to add a warning page in http mode, or a few lines to the documentation about https, anything that affects the code itself.

As a general rule, I still think that actual feature requests should be either kept open and flagged as such or closed as wontfix/rejected, rather than closed as "we may consider this but it is not a bug". Beyond the development of EmonCMS, you guys are doing a great job sorting the bugs in the bugtracker. Congratulations for this, really. You've lowered the number of open bugs to 25, still counting. It is great to aim at zero open bug. But I believe a feature or enhancement request doesn't count as a bug and does not pollute the bugtracker if properly flagged/labeled. Just my opinion. These are workflow choices. There is no absolute truth.

Have fun for the last few hours of 2015.

jpsingleton, 2016-01-05:
Configuration can (and should) be code. This way it is repeatable and consistent. See the shell script mentioned above. A script could be included in the repository that allowed the user to easily set up SSL. Or use something like Chef or Puppet.

I'm happy to close this issue only once enabling SSL has been tracked elsewhere. It's too important to get lost. It may have once been acceptable to not use SSL but these days it is not optional for any site which involves user authentication (in this case emoncms.org and openenergymonitor.org forums).

It's so cheap (or free) and easy that I use it even for static sites. This proves to the user that they are connecting to the site that they think they are. It stops malicious injection and redirects. SSL/TLS is not just for encryption, it is for validation as well.

If you would like some help setting up SSL with CloudFlare (for free) on emoncms.org and openenergymonitor.org then give me a shout. It's really quick and simple.

hockeyshooter, 2016-02-16:
FWIW we've just had to add an SSL certificate to our own-hosted emoncms and all I had to do was tell the data contributors to change the URL from http to https in their emon Pis and its working fine.

jpsingleton, 2016-02-16:
The Android app already has the option for SSL too. The only change would be removing the [text](https://github.com/emoncms/AndroidApp/blob/06a0c9fd441c5ac34abe85846d2064d5d008338f/app/src/main/res/xml/main_preferences.xml#L21) that says emoncms.org doesn't support HTTPS. It could even auto set HTTPS on if emoncms.org is used.

Let's Encrypt instructions are [here](https://letsencrypt.org/howitworks/) and [here](https://github.com/letsencrypt/letsencrypt/blob/master/README.rst). Even the old way is pretty simple, just remember not to use SHA1 signed certificates any more as they are deprecated.

TrystanLea, 2016-02-25:
Pleased to say emoncms.org now supports HTTPS, thanks for the encouragement. 

By default: accessing emoncms.org via the home page or login page or any other non-authenticated page will automatically redirect to https.

For existing devices posting to http, I have not implemented an automatic redirect, because I would like to monitor the effect on server load as users switch. @jpsingleton do you have experience of the implications of the extra overhead? 

The option probably needs to be there to use http for bandwidth constrained connections, a https request consumes about ~3.3kb vs 0.9kb for http and the connection is not persisted if posting from emonhub. 

Thanks again

jpsingleton, 2016-02-26:
Good work! You get an [A](https://www.ssllabs.com/ssltest/analyze.html?d=emoncms.org). Better than a lot of people. Let's Encrypt certificates have a short validity period but if you're using their auto-renew process then this if fine and should take care of itself.

In my experience, if configured correctly then the extra server load should be negligible. I'd be interested to see what you experience though. My browser selected a Diffie-Hellman Elliptic Curve key exchange which is fairly easy to compute. In fact I've just been writing about this for my new book.

Redirecting devices to HTTPS may not be secure as if the data is posted in the initial message then it is already sent in the clear before the redirect. You can use HSTS headers but you need to be careful with these. It's probably better to simply encourage users to switch to HTTPS and add this as a default to a future release of emonhub and the native app.

If bandwidth is a major concern then MQTT may be a better choice than HTTP. There are performance techniques for HTTPS that involve reusing an existing connection too.

You're unlikely to have issues with older software not supporting SNI as you're on a dedicated host with it's own IP address. Worth keeping an eye on though.

closed 2016-02-26 by jpsingleton

TrystanLea, 2016-03-02:
Thanks! The load seems to be ok so far but then the traffic on https is still limited. Good point about redirecting. Yes Id like to investigate using MQTT.
Thanks again for all of your help!

jpsingleton, 2016-03-03:
No problem! Glad to be of help.

TrystanLea, 2016-03-07:
I wonder if I could get your input @jpsingleton on another part of the security of emoncms.org. I've published the latest build of emoncms.org here https://github.com/emoncms/emoncmsorg and a key part is that there is a storage server and main server with communication between them secured with stunnel. Are you familiar with stunnel? I have documented the configuration briefly here https://github.com/emoncms/emoncmsorg/blob/master/stunnel.md

jpsingleton, 2016-03-07:
I've only used stunnel locally on a pi to enable TLS. So I'm not massively familiar with it.

If your main and storage server are both on the same network behind a firewall then it shouldn't be necessary. The traffic between them will only be on the LAN. For example, you could put them in a Virtual Private Cloud if hosting on AWS. You would firewall them in a classic two tier architecture style.

`Internet -> Public firewall -> Main www server -> Internal firewall -> DB storage server`

I've seen openssl (which is similar) used for redis replication but this was for performance, not security. It compresses the data as a side effect which speeds up data transfer and reduces network saturation.

TrystanLea, 2016-03-07:
Thanks for the quick reply! Yes I had initially assumed that the storage server and main server where on the same LAN in the datacenter but after asking the hosting provider it turns out they are in geographically separate locations and the IP addresses are public of course, they recommended I encrypted. I need to discuss this more with redstation who we host with, it might be good to have both hosted on the same LAN, I need to learn a bit more about the options. We did have a partial outage a couple of weeks back that only affected the storage server network, data was queued on the main server until the network connection that the storage server is on came back - which meant no data was lost.

jpsingleton, 2016-03-07:
Ideally they should be on the same local network, for lots of reasons, not only for security.

A VPN may be a better solution. Even if the traffic is encrypted the DB shouldn't be publicly reachable. Most data stores are not hardened and default configurations often assume a trusted network. At the very least the storage server firewall should block connections not from the main server (or boxes you perform maintenance from). A VPN would be more flexible, especially if you administer from dynamic IP addresses.

TrystanLea, 2016-03-09:
Thanks again, Il discuss this with them and see if they are able to provide this option.

Paul-Reed, 2016-03-10:
James @jpsingleton I'd value you opinion of [Kloudsec](https://kloudsec.com/) as a quick, easy & free way for emoncms users to protect their sites with SSL.  
It appears that Kloudsec obtain a free LetsEncrypt certificate for you, and hold it on their servers, redirecting traffic to your server via https.  They also manage renewal of the 3 month LetsEncrypt certificates, as well as other enhancements.

Paul

jpsingleton, 2016-03-11:
Never heard of them. If you want to trust a CDN with TLS termination then I'd go with CloudFlare, which is also free but more established. You probably shouldn't be giving your master keys to some random startup.

jpsingleton, 2017-07-16:
I know this is an old thread but considering my previous comment I should mention here that there are risks with trusting a CDN. See the [Cloudbleed](https://en.wikipedia.org/wiki/Cloudbleed) incident at CloudFlare from earlier this year.

This shouldn't be an issue here as you're doing your own termination and using Let's Encrypt certs (they will offer [wildcard certs from next year](https://letsencrypt.org/2017/07/06/wildcard-certificates-coming-jan-2018.html) BTW). I used emoncms.org as an example of a site with a good TLS configuration in my [web app book](https://unop.uk/book/). I'm currently writing the [second edition](https://www.packtpub.com/application-development/aspnet-core-2-high-performance-second-edition) and I'm happy to see that you still get an A.