emoncms/emoncms #430: Password security issues

jpsingleton, 2015-12-18:
Please increase the maximum allowed length of passwords from 30 characters to something more reasonable such as 250. The allowed length should also be fed back to the user and could even be used as a maximum length attribute on the form input field. `Password length error` is not as helpful as it could be. I had to dig into [the code](https://github.com/emoncms/emoncms/blob/stable/Modules/user/user_model.php#L170) to find what the restriction was. I also suggest increasing the minimum length from 4 to something more secure such as 10.

As the password is hashed (good work) the length is irrelevant. The hash output will always be the same size. You only need to protect against obscenely large malicious submissions.

Regarding hashes, although a couple of rounds of SHA256 with a salt (i.e. an HMAC) used to be sufficient, these days it can be cracked in a trivial amount of time. It would be better to use something with a higher work factor such as bcrypt or PBKDF2 (with at least 10000 iterations). You can transparently upgrade existing hashes by using them as the input to the new hash algorithm like this pseudo-code: `hash = pbkdf2(hmacsha256(password, salt), salt, 10000)`

elyobelyob, 2015-12-19:
Really? Who would want a password longer than 30 characters?

Jerome-github, 2015-12-19:
Maybe someone who stores his passwords in a safe place rather than remembering all of them.

I don't do that myself, but I agree there is no need limiting to 30. Especially since 

> As the password is hashed (good work) the length is irrelevant. The hash output will always be the same size. You only need to protect against obscenely large malicious submissions.

jpsingleton, 2015-12-20:
Yes, the primary use case is for password managers but also for people who use pass-phrases instead of passwords. I use a tool to generate a long unique password for each sign-up however many people still reuse credentials despite the big risk. This is why a long password and slow hash function are a good idea. 

To turn the argument around, what is the reason for wanting to restrict passwords to a short string?

pb66, 2015-12-20:
Although I would be surprised by a >30 char password, as you say there is no real reason to restrict it to 30 if users want more. But a minimum of ten is a bit extreme, 

I don't recall any other login being quite so demanding, even banking, ebay paypal etc do not demand 10+. Most of my passwords are 7,8 or 9 chars and I struggle to remember them sometimes, 10 is too much, that would force users to use a predictable array of chars (passphase) rather than a slightly smaller more obscure password which is counter productive.

Even raising it to 6 would increase the possible unique permutations over 1600 fold, but I think even that is unnecessary, users are aware better passwords are more secure and can choose a size to suit the application & environment.
Paul

jpsingleton, 2015-12-20:
Have split this into even smaller parts and changed to master branch. https://github.com/emoncms/emoncms/pull/432

elyobelyob, 2015-12-21:
Yes, please don't make the lower limit to 6 or 10. I use less than this, as I am not overly fussed about the security of the information, and I have on a quick and easy username & password.

Thanks

TrystanLea, 2015-12-22:
This is great, thanks for the input. I assume we need to increase the database field length to support 250 characters as well, unless Im mistaken?

TrystanLea, 2015-12-22:
I've just done this in this commit: https://github.com/emoncms/emoncms/commit/8fe4995403911bf71befc6c5f5e73e690050139f

TrystanLea, 2015-12-22:
Realised that the mysql escape string extends a special character containing password beyond the 250 length. Mysql escape should not be needed on a password as it is hashed before placing in the database which should make the sql query safe. I've commented out the mysql escape character lines in this commit: https://github.com/emoncms/emoncms/commit/49d415915b78d5f5535f2a9eeb00443769c9e8ea

There is one issue with this that anyone with passwords containing special characters prior to this update will now not have their passwords modified and so authentication could fail. If this is an issue it might be better to switch back to the old system.. or require that passwords are reset.

Id welcome any thoughts on this.

jpsingleton, 2015-12-22:
You don't need to increase the database column width, although it won't
hurt. A varchar only uses as much space as required.

The password is not stored as plaintext. The output of a hash function is
the same size for any give input. That's the purpose of them.

The SHA256 hash as used produces a 256 bit output. Stored as hexadecimal
these 32 bytes are 64 characters long. This will be the case for all
passwords.

I'd recommend using parameterized queries instead of escape strings. They
are much safer. In particular, the standard email validation check is not
sufficient to prevent a SQL injection. Email should also be escaped but
this will modify some obscure but valid addresses.

I've just done this in this commit: 8fe4995
<https://github.com/emoncms/emoncms/commit/8fe4995403911bf71befc6c5f5e73e690050139f>

TrystanLea, 2015-12-22:
Ah ok, I knew it was not plain text, I have changed it back to 64 bytes https://github.com/emoncms/emoncms/commit/9fbd6ac327de89e9554719d780ae5cafefdcf5fd

TrystanLea, 2015-12-22:
I've changed the register and email update methods to use mysqli prepared statements. I very much appreciate the input checking this. https://github.com/emoncms/emoncms/commit/3189b16dce6d2bbc3997352ba9673cb3d38d0a35

jpsingleton, 2015-12-22:
> statements. I very much appreciate the input checking this. 3189b16

That appears better from just looking at it. If this approach works for you
then I'd recommend using the prepared statements with parameters pattern
consistently throughout the codebase in all SQL operations. Then you don't
need to worry about escaping user input (from a SQL perspective) as the
database won't treat it as a command anyway.

Paul-Reed, 2015-12-25:
Is [this post](http://openenergymonitor.org/emon/node/11867) a result of the change?
Paul

jpsingleton, 2015-12-30:
Looks that way but it appears this has been solved from that thread. Another option to leave the escaping function in but perform the length check before that.

closed 2015-12-30 by jpsingleton