emoncms/emoncms #433: Dashboards Missing

Paul-Reed, 2015-12-21:
If you find that you have recently installed emoncms, or updated your existing installation via git, and noticed that your dashboards have disappeared - [please see this forum post!](http://openenergymonitor.org/emon/node/11798)

Saoclyph, 2015-12-25:
Try updating the database scheme in the admin view, it did the trick for me.
I updated from 8.3.

Paul-Reed, 2015-12-25:
Thanks Saoclyph, however updating the database will not re-install the dashboard module.
It has been removed from the emoncms git core, and needs to be installed as a separate module.

Saoclyph, 2015-12-26:
Ha sorry, I misread "Dashboard missing" with "Dashboard **Widget** missing". I thought wifixcort from the forum thread had posted an issue here.
I had the same issue as him, widget not appearing in the dashboards after updating emoncms core and adding the dashboard module.
I did update the database scheme after updating emoncms core, but I forgot to do it after adding the dashboard module. Updating it later solved the missing dashboard widget.

closed 2017-02-28 by TrystanLea