emoncms/emoncms #437: OpenBem disappeared when upgrading to v9

arthurlutz, 2015-12-30:
I started using openbem https://github.com/emoncms/openbem on v8 (it was shipped with it) and when I upgraded to v9 it has disappeared. 

Is it possible to reinstall it ? Is my data still there ? 

Should the upgrade documentation mention it if you decided not to ship it with emoncms v9 ?

TrystanLea, 2015-12-30:
Hello, yes its possible to keep using it on v9 and your data will still be there. Does it appear under the extras menu for you? is the module still present in the emoncms/Modules folder?

Paul-Reed, 2016-01-03:
@arthurlutz Can you update the thread please.

arthurlutz, 2016-01-04:
Since I followed the upgrade instructions the OpenBem module was not in emoncms/Modules. I copied it from emoncms.old/Modules and it worked out of the box with my data. 

I still think either the module should be shipped with emoncms (+1) or the documentation should add a warning/instructions about the migration of modules.

rossoreed, 2016-01-04:
OpenBem is still there in the emoncms git account, in its own repo, in much the same way as 'app' & 'device'. OpenBem won't be shipped with emoncms as a core module, as we are currently modulising emoncms, and making as many modules optional user options.
However it's good to hear that it works fine with v9 and I'll assign this issue to myself to add OpenBem to the installation options guide.

Paul-Reed, 2016-01-04:
OpenBem is still there in the emoncms git account, in its own repo, in much the same way as 'app' & 'device'. OpenBem won't be shipped with emoncms as a core module, as we are currently modulising emoncms, and making as many modules optional user options.
However it's good to hear that it works fine with v9 and I'll assign this issue to myself to add OpenBem to the installation options guide.

Paul-Reed, 2016-01-06:
OpenBEM now added to the installation options guide on 'master' branch https://github.com/emoncms/emoncms/commit/2b10c3f2a679fab02209586d31b06e3f36ffdca1 and will be migrated to 'stable' in due course.

closed 2016-01-06 by Paul-Reed