emoncms/emoncms #439: Log Level Filtering Is Incorrect

jpsingleton, 2016-01-05:
Whatever the log level is set to full logging occurs. The default of WARN also logs INFO.

It looks like the [original commit](https://github.com/emoncms/emoncms/commit/22d9b8afab417e9110b4207033ec7b302e50f579) is correct but a [later change](https://github.com/emoncms/emoncms/commit/2c466e21c264535a5ae1985de4cf94d98bce8f55) got the comparisons the wrong way round.

I had to disable logging to stop it spamming to the SD card. P.S. Could this be added back into the documentation and updated. It has [gone missing](https://github.com/emoncms/emoncms/commit/818905ba0bb681a724e285d486cedeab4c9f47fa).

Paul-Reed, 2016-01-05:
Hi James, it was removed from the guide because we thought that the new
filtering function was effective, but there's obviously a glitch. I'm going
to cover the new log filtering in the docs shortly.

jpsingleton, 2016-01-05:
Hi Paul, that's great. It would be nice to have it documented.
I've opened https://github.com/emoncms/emoncms/pull/440 to fix the glitch.

Paul-Reed, 2016-01-05:
Merged #440 and log filtering has been re-established.
I'll leave this open until the documentation has been sorted, and assign it to myself.
Thanks

jpsingleton, 2016-01-06:
Thanks Paul. It would be great if this was merged into the stable branch as the logging bug is still present in there.

Paul-Reed, 2016-01-06:
It needs to be signed off by Trystan or Glyn before 'master' & 'stable' are merged - won't be long!

Paul-Reed, 2016-01-06:
Logging options re-added to guide https://github.com/emoncms/emoncms/commit/2b10c3f2a679fab02209586d31b06e3f36ffdca1 in 'master' branch. Will be migrated to 'stable' branch in due course.

closed 2016-01-06 by Paul-Reed