emoncms/emoncms #446: emoncms enable multiuser

kumarmohan7, 2016-01-15:
Hi;

Please enable multiuser , enable user register 

how to control relay through emoncms...?

Paul-Reed, 2016-01-15:
Multiuser is a user option activated by changing the setting in settings.php or are you saying that it doesn't work?

kumarmohan7, 2016-01-16:
ok sir thank you

how to control relay through emoncms...?

Paul-Reed, 2016-01-17:
You can't directly via emoncms, but most users are using additional software, such as node-red or mqttwarn to control a remote relay via MQTT.
Both of them are bundled in the emoncms image file, and node-red comes already installed in the latest Raspbian Jessie OS. The MQTT relay is available from the [OEM shop](http://shop.openenergymonitor.com/three-channel-wifi-mqtt-relay-thermostat/) (when they are in stock!).

closed 2016-01-17 by Paul-Reed