emoncms/emoncms #449: inputs screen blank in EmonCMS 9.3 | 2016.01.16

JumpMaster, 2016-01-17:
After upgrading to EmonCMS 9.3 | 2016.01.16 the inputs page is now blank.  Feeds page has the drop downs by node which is great though.

JumpMaster, 2016-01-17:
I should be more descriptive. The page loads as expected and the message "loading feeds" displays although the feed list is empty. This is also the same in the dashboard list.

Paul-Reed, 2016-01-17:
I can't replicate the blank input/feed pages, they seem fine here, however the new dashboard button does not work, I can't create a new dashboard.

Paul

JumpMaster, 2016-01-17:
That's strange. The new dashboard button works for me and is visible in the dashboard dropdown. Anything I can do to troublshoot?

Paul-Reed, 2016-01-17:
I usually find the browser developers console a good starting point, and see if any errors are reported. Also emonhub & emoncms log files.

glynhudson, 2016-01-17:
I've just updated to 2016.01.16 and the Dashboard option seems to have
disappeared from the Setup menu even though the dashboard module is
installed. I'm pretty sure dashboard were working fine before this update.
I like the drop down feed grouping BTW.

JumpMaster, 2016-01-17:
No errors in the browser console on the dashboard nor inputs.  Also no errors recorded in the emoncms.log.  The log was not been updated although the settings file had it pointed at /var/log/emoncms.log.  After overwriting my settings file it is now in /var/www/html/emoncms.log and updating as expected although there are no errors reported.

Paul-Reed, 2016-01-17:
Yes, unfortunately the log file location was overwritten in the latest
merge. It will be moved back to /var/log shortly.

pb66, 2016-01-17:
I've just "fully" updated my emonpi and when I read this thread I looked for these issues and found none of them. Realizing this was due to being on the released or "stable" branch I tried to switch to "master". I had some issues due to the scripts/logger folder and all it's contents being root:root and wasn't able to update the "master" until I deleted the folder. 

Once I could update master I found all the same symptoms described above except the original "empty inputs or feeds page" which I will come back to at the end.

After wrestling with this for a while I realised the dashboard folder was almost empty, just a .git folder and very little else. I deleted and re cloned the dashboard folder and I regained the "dashboard" option in the dropdown.

At this point I recognized the issue with the "new dashboards" button not working and no dashboards listed. Back in July when we discussed the group_collapse I brought this issue up on the emoncms 9.n thread. It is because there is no grouping for dashboards, so they are not listed they are lost. the button is working but each time you press it a new dashboard with no name or group is created. If you switch back to stable branch you will see them all.

so 
@Glyn - you need to re-clone dashboards
@Paul - you didn't see the dropdown issue as you are not using the emonpi updating routine, but if you switch brabches you will see all the new dashboards you have created :-)
and
@JumpMaster - I'm guessing you either created your feeds and inputs in a previous version of emoncms and carried your data over or you have deleted the grouping "tags" since they were created with default tags. You cannot see the inputs and feeds as they have slipped through the same crack as the dashboards. If you switch to "master" and tag all your feeds and inputs you should see them when you switch back to master.

I really like the way the grouping and collapsed groups work too, so I'm not picking fault at all, but there needs to be something like a default "no tag defined" group so that non-grouped members of any page do not disappear if they have no tag.

I would also like to see dashboards grouped and there are several ways that could be done, It would be nice to discuss further.
 I have previously suggested sub-grouping pages like a webpage/folder relationship so that groups can be on the main menu bar and dropdown to reveal pages, this reduces the items needed on the menubar.
Alternatively I also suggested grouping by published/public/private etc
But the dashboard grouping is not urgent as if the "safety net" of an "untagged" group is implemented across the board the dashboards will become accessible again
 
Paul

pb66, 2016-01-17:
Attached my console output incase you wish to delve deeper into the master<>dashboards cloning issue
[updating_master.txt](https://github.com/emoncms/emoncms/files/93509/updating_master.txt)
Paul

pb66, 2016-01-17:
@JumpMaster - I take back the suggestion the inputs are ungrouped as the node id *IS* the group so that cannot be the case. But I do recall when I previously tried this my data exported from emoncms.org didn't show at first, but I cannot recall why. Are your inputs from an older install version? and if you use the input api to create a new input is that visable?
as yet I'm not able to replicate the unpopulated inputs page .
What version did you upgrade from have you used admin => update database ?

JumpMaster, 2016-01-17:
Hi Paul,

I upgraded from 2015.12.30. Most of my feeds have been running since version 8.3 and a few were created two days ago. My data has never been exported and imported.

I'm running php 5.6.17 and MySQL 5.1.3 in case there's anything obviously wrong there...

Thanks!

JumpMaster, 2016-01-17:
I just tried adding a new input as node 2 with the below request.

/input/post.json?apikey=myapikey&node=2&json={kevtest:3063}

All of my inputs now appear but if I delete "kevtest" within node 2 the inputs within node 2 disappear.

Paul-Reed, 2016-01-17:
@pb66 - Paul, I've seen your console output, but I can't see why the permissions would be root:root in /scripts/logger/ because the installation script [specifically chown's the var/www directory to be owned by the user](https://github.com/openenergymonitor/emonpi/blob/master/emoncms9install#L86) (unless the script was run whilst the user was su?) In any event, why just /scripts/logger and not the whole emoncms directory??  

Paul

glynhudson, 2016-01-17:
Thanks Paul, re-cloning the dashboards worked. I've now got the dashboards
back in the Setup tab and able to replicate the issue of not able to create
a new dashboard (+New) button does not do anything.

If these changes were to be merged into Stable branch which would mean they
would be pushed out to all emonPi users running the new emoncms V9 (Dec22)
SD card image if they need to also re-clone the dashboards module this will
be a real issue. Most emonPi users will be unable to ssh in and re-clone.
Is there a way this update could be managed without a re-clone?

Paul-Reed, 2016-01-17:
@glynhudson - Glyn, would it be a problem at all? Hasn't the dashboard module disappeared because you've switched branches?  
If master & stable were merged, users would just update and surely their existing dashboard module would remain intact??

pb66, 2016-01-17:
@Paul-Reed - I have no idea !! that's why I included the console output. I took a freshly downloaded 22dec image, it update at boot on the 8th Jan, I don't recall updating since then but cannot rule it out, until today I ran the updater via the emoncms "update emonpi" button and voila, the log shows the rest.

Here are the previous "before and after" logs for today's automated update before the manual branch switching (checkouts), pulls and cloning. They don't seem to object to the logger folders permissions so I would be inclined to think the permissions are wrong in the repo and are then over ruled by your chown command IF done manually not via emonpi script.

[20160115_update.txt](https://github.com/emoncms/emoncms/files/93589/20160115_update.txt)
[20160108_update.txt](https://github.com/emoncms/emoncms/files/93590/20160108_update.txt)

@glyn - I think the button is working you just cannot see all the new dashboards you are creating because they are created without a group tag and therefore not displayed, you can confirm by switching back to "master" or use phpmyadmin to take a peek at the mySQL tables.

I think you will need to test the update route further, but I believe the issue here is not the updates that will find there way into the "stable" branch but the fact you are using the "master" branch to view them so only the emonPi users that use the master branch are affected not the stable users. but please try that from a fresh image soon after these changes are merged to be sure.

The file and folder permissions should be reviewed through out the repo, there is for example no reason to have to set files to executable each time they are downloaded/cloned. Git faithfully preserves the "executable" status, that is why emonhub can be installed in one line, the installer script is already executable before it's cloned. clone and run that's all that should be required no chown's or chmod'ing.

Paul

pb66, 2016-01-17:
Just took a closer look at the previous logs and you can see on the 8th jan the logger files and folders had their "mode" permissions altered so I suspect someone has made some changes and inadvertantly altered the permissions when committing the changes.

I expect the reason the chmod in the install guide didn't work is because emoncms was originally installed to the image before the 22dec and does not get re-run, the install scripts in emonPi are pretty redundant unless creating a new SD image from scratch.

Paul

Paul-Reed, 2016-01-17:
@pb66 - Paul, the logger files and folders were only committed to emoncms on the 2nd Jan, so they wouldn't have been included in the 22 Dec image. Your update on the 8th Jan appears to show the files/folders being added - not updated. Checking back through git file history, no other changes have been made to them since that initial commit.
Are you sure that git retains file permissions, as I can only find reference that it retains the executable file status (which I wasn't aware of anyway - thanks!).

glynhudson, 2016-01-17:
@pb66 Ah good point, the dashboard module disappearing was due to checking out Master from Stable. Once changes are merged into stable then when user updates dashboards will stay intact.

Since I am on master I should be able to see the dashboard I created using the +New button. It seems the newly created dashboards are appearing on the top bar but not being listed on the dashboard page, see screen grab: 

[image]

pb66, 2016-01-17:
As far as I'm aware the permissions are retained but not the ownership, the numbers you see eg 100644 are permissions the "644" is owner R&W, group R, all users R, who ever clones the repo becomes the owner and the group is set to the owner too.  

*"the logger files and folders were only committed to emoncms on the 2nd Jan, so they wouldn't have been included in the 22 Dec image"*
exactly, so they were not installed using the "clone repo" routine in the emonpi repo so the "chown pi:pi" was never run on them. 

Why or how they are root:root I cannot figure out I have just looked through the emonpi repo but it's a bit of a maze and you cannot search a forked repo so search won't work until Glyn updates or severs his personal repo. The use of chmod and chown is very frequent throughout the install and update routines so it could be an error, but I cannot find it.

@glynhudson - *"Since I am on master I should be able to see the dashboard I created using the +New button."* No, because dashboards have no tag (group) to be grouped by. Your dashboard page shows all your dashboards by the group they are in. All your dashboards have no group, therefore they are not there. Nobody using the "master" repo will see the dashboards listed on the dashboards page.

chaveiro, 2016-01-18:
See if  9.3 | 2016.01.18 fixes any of this issues.

JumpMaster, 2016-01-18:
Just like magic all problems have been fixed :+1: Thanks @chaveiro

glyn, 2016-01-18:
@pb66 - please note that Glyn Hudson is @glynhudson. :-)

pb66, 2016-01-18:
Ooops!1 My apologies @glyn.

pb66, 2016-01-18:
All good here too ! 
Thank you @chaveiro.

Paul-Reed, 2016-01-18:
If you haven't already created any dashboards, you are unable to create one (as mentioned above).  
With Dashboard module installed, Setup > Dashboards brings you to the Dashboard list (which is naturally empty). Clicking  '+ New' does nothing whatsoever, the screen remains unaltered.  
Clean install v9.3 - 2016.01.18

Paul-Reed, 2016-01-18:
Doh! forgot to update the database - sorry!
Now working OK.

closed 2016-01-18 by Paul-Reed