emoncms/emoncms #452: MQTT Inputs fail on multiple nodes

tgmaxx, 2016-01-18:
I used MQTT inputs to a single node number successfully on my local emoncms stable branch 9.3. I am using Node-Red to output data to Emoncms via MQTT. I use MySQL timeseries for all Emoncms storage. I am using redis.

But, when I created a second node, strange things began to happen. I had previously deleted MQTT inputs being created on their own. If I deleted them, they would pop-up again. And, data in the first node was being changed to incorrect values. After a few hours trying to figure out what was happening, my Input table apparently became corrupt (?) and all my Input processing on the Input screen showed "error". I turned off "mqtt_input", rebooted and deleted all MQTT inputs. Then I did a repair on my database tables in phpMyAdmin. Finally I was able to get my Input Processing icons back.

I changed to using the emoncms node in Node-Red instead of MQTT and everything is fine now. But, I wanted to report that there is an issue with MQTT inputs, apparently when sending more than one Node number.

Paul-Reed, 2016-01-18:
This could be more of a configuration issue, than a an issue with emoncms.
When you set up your mqtt node in node-red, did you at any time select Retain = True?
If so node-red will set it's retain flag, and will continually try to deliver 'undelivered' messages, despite the node being deleted, or system reboots. [It is discussed here](http://www.hivemq.com/blog/mqtt-essentials-part-8-retained-messages) and the way to resolve the issue.  
Could this be the issue?  
Paul

tgmaxx, 2016-01-18:
Yes, I did use Retain=True when I initially setup my flow. But, I had deleted those MQTT outputs (with the retain) before I started seeing the problem. I guess that didn't matter that they were deleted -- my previously deleted inputs were mysteriously popping back up, exactly as your link describes.

Anyway, through all this, I found out that the emoncms nodes available for Node-Red are an easy way to send data. Thanks.
Tom

closed 2016-01-18 by Paul-Reed