emoncms/emoncms #472: MQTT needs multiuser support with authentication

chaveiro, 2016-02-13:
On https://github.com/emoncms/emoncms/pull/471 i reverted the user id key location from settings.php to the phpmqtt_input.php script file because it's not related to emoncms instalation but to the instance of that script running for a particular user.
Best is to make the phpmqtt_input.php script generic for multiuser support, this will require mqtt autentication. Use of apikey or device key?
What kind of auth is used usualy for mqtt across the internet?

glynhudson, 2016-02-13:
Good idea. I'm happy with that change of user ID location. Authentication with MQTT is usually done at server level i.e authentication to MQTT server. Sending API key in the topic sounds like a reasonable way to achieve what we want.  We would like to add MQTT support to Emoncms.org in the future. This could be one way to do it. I will discuss with @TrystanLea

Paul-Reed, 2016-02-13:
[This article by HiveHQ](http://www.hivemq.com/blog/mqtt-security-fundamentals-payload-encryption) seems to describe a typical model.

nachoferre, 2016-04-11:
Hi im currentyly trying to connect my mosquitto broker with emoncms but i dont recieve the data. I know its because of the apikey. 
could you provide an example of use?, because i have tried the one above but im not sure i did it right.

I have tried like this: basetopic/[key]/0/1
if i use the pipe "|" the broker complains.
Thanks in advance

glynhudson, 2016-04-11:
See here for standard mqtt api users on emonBase and emonPi by emonhub

https://github.com/emoncms/emoncms/blob/master/docs/RaspberryPi/MQTT.md

- sent from my mobile device

nachoferre, 2016-04-12:
Thanks!
I was able to make a connection in one way, emoncms receiving. For what i have read u need emon hub in order to be able to send mqtt messages from emoncms. Am i right?

glynhudson, 2016-04-14:
No, you can publish to mqtt without emonhub. Just using emoncms publish to
mqtt input process. Ensure mqtt_input services ks running and you get mqtt
connected message in the log. Base topic and authentication details are set
in settings.php

- sent from my mobile device

closed 2017-02-28 by TrystanLea