emoncms/emoncms #480: Problems pecl install on a fresh xenial

jhagberg, 2016-02-21:
On a fresh Ubuntu Xenial install I needed to do the following steps to be able to run
pecl install channel://pecl.php.net/dio-0.0.6 redis swift/swift

rm /usr/lib/php5/build/ltmain.sh
ln -s /usr/share/libtool/build-aux/ltmain.sh /usr/lib/php5/build/ltmain.sh

The symlinc is non existing. 
This is when following the install linux guide.

glynhudson, 2016-02-21:
Thanks for the info. Do you think this may change between now and the end of April when Xenial is due to be released? It sounds like a bug in Xenial rather than an Emoncms required pecl install step.

jhagberg, 2016-02-23:
Yes this is defiantly a Ubuntu bug and not emoncms. But I wanted to report it here if someone else hits the bug.

closed 2017-02-28 by TrystanLea