emoncms/emoncms #486: API list available without logging in

glynhudson, 2016-03-03:
Currently the API's for Emoncms are not documented anywhere online accept in the application itself. Potential users should be able to view the Emoncms Input and Feed APi's without having to create an account to allow them to properly evaluate the Emoncms platform. 

I would like to hear what you think would be the best way to maintain a documented API list and were such a list should be hosted? On the Emoncms GitHub Wiki? Or in an api.md file in the Emoncms repo? It would be great to somehow link this API page to the actual API list within Emoncms to ensure it does not quickly  become outdated.

elyobelyob, 2016-03-04:
Not sure if this is a decent pointer ...
http://owntracks.org/booklet/features/location/

At the footer it says "
Built with MkDocs <http://www.mkdocs.org/> using a theme
<https://github.com/snide/sphinx_rtd_theme> provided by Read the Docs
<https://readthedocs.org/>."

Seems fairly complete and could be a start.

chaveiro, 2016-03-29:
Keep documentation close to the source code is the best approach.
What's wrong with the current documentation pages?

glynhudson, 2016-03-29:
User needs to be logged in to view the API docs. This is fine for current
uses but seems rather obscure to new users who want to evaluate the
applicant API without creating an account.

It seems a shame to make new users jump the hoop of creating an account
just to evaluate the api.

Separate topic but I would also like to make a demo emoncms account with a
demo dashboard(s) and feeds for new users to evaluate. I imagine a
dashboard showing every type of vis emoncms currently supports as live
showcase.

I have just been thinking of the sort of things I like to see available
when evaluating a new Web application platform

- sent from my mobile device

glynhudson, 2016-04-19:
Thanks to @TrystanLea API docs is now accessible when logged out of Emoncms. If you are logged in then your API key will automatically be populated :+1: 

https://emoncms.org/site/api#input
https://emoncms.org/site/api#feed

closed 2016-04-19 by glynhudson