emoncms/emoncms #487: Blocking with the virtual feeds

RDCH106, 2016-03-04:
When in process list of an input the value is operated with the operator `*feed` and the feed is virtual, the following process lists of the following inputs that have a higher id, are not processed.

The virtual feed used has its own process list and it is a RealTime data type. It works correctly, but when the feed is used in a input process list the problem appears.

chaveiro, 2016-03-07:
A few print screens of the procceslists would be helpful to understand the case.
Try use the operator '* source feed' instead of '* feed' to process virtual feeds.

RDCH106, 2016-03-08:
Thanks for your answer!

First of all I can not use your suggestion because `* source feed` is **only available in the feed panel**. I have the problem in the inputs panel bacause when I use the avaiable operator `* feed ` with a virtual feed, the process list of the inputs behind the input are not processed.

I take snapshots of the behaviour when I use the virtual feed in the process list. The first one is the conflitive process list.

[image]

The second is the view of the inputs. All the inputs have recent values.

[image]

The third is the view of the feeds. All the the feed calculated after the conflictive process list have stopped to update the values.

[image]

The virtual feed used is a boolean (0 or 1) value that indicates if the system is running when there is power in the monitorized system. The feed is saw in the fourth snapshot:

[image]

If I use a normal feed the problem is not present, but I am saving not useless historical information. I need check the instant derived value saved in "Running" feed. Depends on the value of "Running", the results of the operations of the input process list are absolute zero or the value calculated.

chaveiro, 2016-03-08:
I cant see anything obvious. Recommend a few tests:
1 - Are you sure it's not a /feed dividing by zero ?
2 - Does any error appear on the logs?
3 - Try to post data like your device would, but from the browser and see if you get an OK or some php error.

RDCH106, 2016-03-10:
Test results:

1 - When the process list is executed with a no virtual feed, the problem does not appear. If it would be a division by zero, the problem will be reproduced in the no virtual feed.

2 - There is the following errors repeated for each call in the process list during the execution:

```
2016-03-10 09:53:25.503|ERROR|process_model.php|__call() Call to undefined module method. Missing function on module? method=source_feed_data_time 
2016-03-10 09:53:28.760|ERROR|index.php|Invalid API key 'xxxxxxxxxxxxxxxxxxx'
```
Some idea where I must start to research?

3 - If the inputs are fed from the browser, the errors are the same as in the second point.

chaveiro, 2016-03-29:
That log give an error but i can't reproduce. Do you have any custom changes to emoncms processes code?

RDCH106, 2016-04-04:
@chaveiro The only custom change is `Reset to One`:

[https://github.com/emoncms/emoncms/pull/484/files/1230b5980aadb64aba04fb90498d859ed35e65ef#r54187834](https://github.com/emoncms/emoncms/pull/484/files/1230b5980aadb64aba04fb90498d859ed35e65ef#r54187834)

The change was proposed with others in this Pull Request, but finally refused:

[https://github.com/emoncms/emoncms/pull/484](https://github.com/emoncms/emoncms/pull/484)

I think that the lines added in `process_processlist.php` are not the reason of  the bug. In the Open Energy Monitor Forum, other users have reported the same problem:

[https://openenergymonitor.org/emon/node/10977](https://openenergymonitor.org/emon/node/10977)

I will continue investigating the problem.

TrystanLea, 2017-02-28:
Is this still an issue? Im not familiar with virtual feeds unfortunately to comment or help with this.

RDCH106, 2017-03-01:
@TrystanLea I don't know... I will check again and I tell you.

RDCH106, 2017-05-23:
Since EmonCM v9.7.3 (2016.09.28) it seems the problem is not present.

@TrystanLea We can mark it as resolved!

closed 2017-05-23 by Paul-Reed