emoncms/emoncms #489: If emoncms is installed in subfolder of the domain, set cookie path

arthurlutz, 2016-03-14:
The cookie set by emoncms does not seem to use a path restriction. I have an install in sub.domain.org/emoncms/ it would be nice for the emoncms cookie information to be restricted to that subfolder.

gablau, 2016-03-15:
I've also found this problem! I have two emoncms installed in two different folders (eg. /emoncms and another in /emoncms9)

I resolved it this way:
on file: https://github.com/emoncms/emoncms/blob/master/Modules/user/user_model.php
before `session_start();` al line 101 put this line:
`$cookie_params=session_get_cookie_params();
session_name('EMONCMS_SESSID'); //name of cookie 
$cookie_params['path']=dirname($_SERVER['SCRIPT_NAME'])."/"; //get subdir installation
$cookie_params['httponly']=1; //not pass cookie to javascript
session_set_cookie_params($cookie_params['lifetime'], $cookie_params['path'], cookie_params['domain'], $cookie_params['secure'], $cookie_params['httponly'] );`

I hope I was helpful

emrysr, 2018-07-12:
Successfully tested on a raspberry pi by temporally switching branches:

```
git remote add dev http://github.com/emrysr/emoncms.git
git pull dev
git checkout dev/feature/issue-489-user-session-settings
```

## Test
This was not possible on the current stable branch. Works fine under the dev/feature/issue-489-user-session-settings branch.

Proves that the session variable is shared to other installations:

1. create "copy": `sudo ln -s /var/www/emoncms /var/www/html/emoncms/emoncms-copy`
1. log in to http://emonpi.local/emoncms
1. log in to http://emonpi.local/emoncms/emoncms-copy
1. log out of http://emonpi.local/emoncms
1. if the 2 versions share the session data, you'll be logged out of 
http://emonpi.local/emoncms/emoncms-copy

**don't forget to remove the test symbolic link and switch back to the stable branch** 
- `sudo rm /var/www/html/emoncms/emoncms-copy`
- `git checkout origin/stable`

TrystanLea, 2018-07-12:
Great thankyou @emrysr, tested and merged

closed 2018-07-12 by TrystanLea