emoncms/emoncms #493: in Sub Domain Realtime graph not comming

kumarmohan7, 2016-03-17:
Hi ;

emoncms sub domain real time graph is not coming error is"URL Not acceptable"

with sub domain address

http://iot.smartiot.co.in/kumarmohan7

[image]

with out sub domain address

http://smartiot.co.in/iot/kumarmohan7

[image]

Please Help 
how to salve this problem....?

Thank You.

kumarmohan7, 2016-03-17:
Hi ;

emoncms sub domain real time graph is not coming error is"URL Not acceptable"

with sub domain address

http://iot.smartiot.co.in/kumarmohan7

[image]

with out sub domain address

http://smartiot.co.in/iot/kumarmohan7

[image]

Please Help 
how to salve this problem....?

Thank You.

chaveiro, 2016-03-29:
if you edit the dashboard does this graph appears?

closed 2017-02-28 by TrystanLea