emoncms/emoncms #494: Cant add data in imported feed

jhagberg, 2016-03-21:
I have a problem with imported feeds from another installation. I cant add data in the feed. The old data is there and I can add data in newly created feeds but not in the imported feeds. 
Any tips on this?

glynhudson, 2016-03-23:
What error do you see when trying to add data to imported feeds? Any output in developer console?

I was having the opposit error, I could add data to the imported feeds but I could not create new feeds:

[image]

This has now been fixed with: https://github.com/emoncms/backup/commit/d38901f919d0ca19e3400b10c2af1c1aaa7b1258

Please post the output of `ls -la data/phpfina/`
To fix permissions for ready imported data run: 

```
$ sudo chown www-data:root /home/pi/data{phpfina,phptimeseries} 
$ sudo chown -R www-data:root /home/pi/data{phpfina,phptimeseries} 
```

closed 2016-03-30 by glynhudson

jhagberg, 2016-03-30:
Yes you are correct.
phpfina$ ls -lha
total 12M
drwxr-xr-x  2 www-data root       32 Mar 22 08:55 .
drwxr-xr-x 51 root     root       51 Mar 30 08:14 ..
-rw-r--r--  1 root     root     654K Feb 21 12:01 17.dat
-rw-r--r--  1 root     root       16 Feb 21 12:01 17.meta
-rw-r--r--  1 root     root     654K Feb 21 12:01 18.dat
-rw-r--r--  1 root     root       16 Feb 21 12:01 18.meta
-rw-r--r--  1 root     root     654K Feb 21 12:01 19.dat
-rw-r--r--  1 root     root       16 Feb 21 12:01 19.meta
-rw-r--r--  1 root     root     654K Feb 21 12:01 20.dat
-rw-r--r--  1 root     root       16 Feb 21 12:01 20.meta
-rw-r--r--  1 root     root     3.9M Feb 21 12:01 21.dat
-rw-r--r--  1 root     root       16 Feb 21 12:01 21.meta
-rw-r--r--  1 root     root     654K Feb 21 12:01 26.dat
-rw-r--r--  1 root     root       16 Feb 21 12:01 26.meta
-rw-r--r--  1 root     root     9.2M Feb 21 12:01 4.dat
-rw-r--r--  1 root     root       16 Feb 21 12:01 4.meta
-rw-r--r--  1 root     root     9.2M Feb 21 12:01 5.dat
-rw-r--r--  1 root     root       16 Feb 21 12:01 5.meta
-rw-r--r--  1 www-data www-data  46K Mar 30 13:39 67.dat
-rw-r--r--  1 www-data www-data   16 Mar 22 08:53 67.meta
-rw-r--r--  1 www-data www-data   12 Mar 22 08:55 69.dat
-rw-r--r--  1 www-data www-data   16 Mar 22 08:53 69.meta
-rw-r--r--  1 root     root     9.2M Feb 21 12:01 6.dat
-rw-r--r--  1 root     root       16 Feb 21 12:01 6.meta
-rw-r--r--  1 www-data www-data  46K Mar 30 13:39 70.dat
-rw-r--r--  1 www-data www-data   16 Mar 22 08:53 70.meta
-rw-r--r--  1 www-data www-data  46K Mar 30 13:39 77.dat
-rw-r--r--  1 www-data www-data   16 Mar 22 08:55 77.meta
-rw-r--r--  1 www-data www-data  46K Mar 30 13:39 79.dat
-rw-r--r--  1 www-data www-data   16 Mar 22 08:55 79.meta
-rw-r--r--  1 www-data www-data  46K Mar 30 13:39 80.dat
-rw-r--r--  1 www-data www-data   16 Mar 22 08:56 80.meta

jhagberg, 2016-03-30:
Any tips on merge the data in the differents feeds?

glynhudson, 2016-04-02:
I see some of your feeds data is owned by root user it all should be www-data. Fix with:

```
$ sudo chown www-data:root /home/pi/data{phpfina,phptimeseries} 
$ sudo chown -R www-data:root /home/pi/data{phpfina,phptimeseries} 
```