emoncms/emoncms #495: Windows install instructions error

AndyRap, 2016-03-22:
Hi,

Found one slight error in the Windows install instructions.  The line that reads -

$log_filename = dirname(__FILE__).'/' . 'emoncms.log';

should read -

$log_filename = dirname(__FILE__).'\\' . 'emoncms.log';

Keep up the great work :)

Regards
Andy Hodges

glynhudson, 2016-03-22:
Thanks, been fixed.

https://github.com/emoncms/emoncms/commit/e8866a206611572db4f4c5e1603832095eaf2420

AndyRap, 2016-03-22:
Sorry Glyn,

I should have double checked my original post :/

The correct line should have a double backslash in it.........

$log_filename = dirname(FILE).'\\' . 'emoncms.log';

I promise it's right this time

Regards
Andy Hodges

glynhudson, 2016-03-22:
Is this correct, I can only see a single backslash in your post:

$log_filename = dirname(FILE).'\' . 'emoncms.log';

AndyRap, 2016-03-22:
Well, kind of......

GitHub seems to remove a backslash from the text when you hit the comment button.  Try it yourself.

So, in order to display the correct line I need to put three backslashes in, so that it shows two?!?

The correct line should look like this when viewed -

$log_filename = dirname(FILE).'\\\' . 'emoncms.log';

Regards
Andy Hodges

glynhudson, 2016-03-22:
Ah yes, I see! How strange. I've new added double post to the guide:
https://github.com/emoncms/emoncms/commit/0da11d2f056040cc2f284fa0a6d12f65effaee94

Thanks

closed 2016-03-22 by glynhudson