emoncms/emoncms #496: Application crash

adfaure, 2016-03-22:
Hi ! 
I wanted to use emoncms for my project. 
I played with emoncms (on my raspberypi)  a moment before it tottaly emoncms crashed down. 

I can acces to the application but it seems some pages failed to load. 
* When i try to go to the tabs feeds;
* When I try to go to the tabs visualization.

Into the developpers console I saw the request which failed : 
`http://emon/emoncms/feed/list.json`

adfaure, 2016-03-22:
I dropped the database an re-create it, it works now.

AndyRap, 2016-03-23:
I have a similar issue and it has occurred a number of times, each has required a rebuild of the Pi and EmonCMS along with it.

It has just happened again after running without issue for a couple of days.

My inputs update their values but with a "please wait....."  in the feed processing section

[image]

and if you look at the feeds screen it just has "loading..." and nothing else

[image]

Any help would be appreciated.

Regards

Andy Hodges

Paul-Reed, 2016-03-23:
What makes you think that it's the same problem @AndyRap ?

adfaure, 2016-03-23:
I can confirm, I had the same screen :)

AndyRap, 2016-03-23:
Paul, 

20+ years as an IT consultant and a bit of gut feeling;  The description of the feeds page not loading sounded exact and I do recall at one point (previous crash, days ago) the vis module did not load for me, IIRC I got a browser error rather than any php error, like a 404 but not.  Sorry to be vague but I didn't record the details.

As this issue has happened maybe 4 times in the past week I created an SD image of the EmonCMS install but before any inputs or feeds have been entered.  I have just reimaged the SD card again so I can do no further testing at this point on a 'failed' install.  All previous installs have had the rootfs moved from the SD onto a Sandisk 16GB USB stick, using the supplied script.  This time I will leave it on the SD for a week and see if it occurs in SD configuration.

I had thought it was something to do with trying to use the 'app' section but I can disprove this as the last install had this working a treat.  I also though, and still do to some extent, that it may have something to do with creating a histogram feed.  Although all the other times the crash had happened when playing around with creating feeds on a single node, this time I had setup a histogram feed on node 1, hence my uncertainty, and was trying to configure a histogram on node 2 when the failure occurred.  Node 2 has only been added this afternoon.

If there is any more information that you think I may be able to give you to help please ask.  I will try to keep better, step-by-step, notes this time so I can try and pin point what series of events leads to the error.

Next time it occurs I will save an image of the SD so the devs can download and investigate if they so desire.

Also, as I have only be using EmonCMS for the past week and came straight in at 9.4 so I can not state if the issue existed in any previous version.

Kind Regards
Andy Hodges

glynhudson, 2016-03-23:
Thanks for investigating guys, please let us know if you find the cause for this bug

chaveiro, 2016-03-23:
The error describes that apache crashed with a segmentation fault. That is usually related to hardware problems.

glynhudson, 2016-03-24:
Are you running on a Raspberry Pi? I had issue with Segmentation fault due to php5-redis issue: 

From:
https://github.com/openenergymonitor/emonpi/blob/master/docs/SD-card-build.md#7-install-emoncms-v9-core-stable-branch

> Note: at time of writing the version of php5-redis included in the Raspbian Jessie sources (2.2.5-1) caused Apache to crash (segmentation errrors in Apache error log). Installing the latest stable version (2.2.7) of php5-redis from github fixed the issue. This step probably won't be required in the future when the updated version of php5-redis makes it's way into the sources. The check the version in the sources: sudo apt-cache show php5-redis

To fix:
```
git clone --branch 2.2.7 https://github.com/phpredis/phpredis
cd phpredis
(check the version we are about to install:)
​cat php_redis.h | grep VERSION
phpize
./configure 
sudo make 
sudo make install
```

adfaure, 2016-03-24:
Hi, 
problems solved for now. I'll try to use emoncms  a moment to confirm it :) 
Thanks.

glynhudson, 2016-03-24:
RasPi install guide has been updated. Could you confirm version of php5-redis currently in the latest Wheezy sources (after updating) 

sudo apt-get update
sudo apt-get dist-update
sudo apt-cache show php5-redis

Paul-Reed, 2016-03-24:
I've just checked Jessie and it's Version: 2.2.5-1

Paul

adfaure, 2016-03-24:
Did you mean : `dist-upgrade` ?

adfaure, 2016-03-24:
```
$ sudo apt-cache show php5-redis
Package: php5-redis
Source: php-redis
Version: 2.2.5-1
Architecture: armhf
Maintainer: Debian PHP PECL Maintainers <pkg-php-pecl@lists.alioth.debian.org>
Installed-Size: 248
Depends: libc6 (>= 2.4), phpapi-20131226, php5-common (>= 5.2.0), php5-common (<< 6.0.0)
Suggests: redis-server
Homepage: http://pecl.php.net/package/redis
Priority: optional
Section: php
Filename: pool/main/p/php-redis/php5-redis_2.2.5-1_armhf.deb
Size: 98268
SHA256: aa9bf469310ca2fde6f472432b8a0a2de111ac688430293f1ca16797c2272f4f
SHA1: c4bcf30d76cc3997d8c7d3437705a207412ef430
MD5sum: 79fe2686b8cdecbbbb32cc143dd7ab8a
Description: PHP extension for interfacing with Redis
 This extension allows php applications to communicate with the Redis
 persistent key-value store. The php-redis module provides an easy object
 oriented interface.
Description-md5: 3d9bd1b1462280436e6583b6eecd249b
```

AndyRap, 2016-03-24:
Same as Paul.  Running Jessie and it's 2.2.5-1.

I haven't update manually since I re-imaged the card so, I can only guess that this is the same version that was running when I had a problem Yesterday.

Currently running again without issue so fingers crossed ;)

Kind Regards
Andy Hodges

AndyRap, 2016-03-24:
Mine has gone again :/

Prior to it stopping I had recently (in addition to many feeds created and running for +12 hrs) added an addition feed to the power input, this was to log to a mysql database.  The additional feed was created without issue and data was logged to mysql for about 5-10 minutes.

I then deleted this feed in the feeds list, then clicked the header link to open inputs, expecting to see an error listed in Node1:Input1: for the feed I had just deleted.

At this point my Input and Feed screens look like the screenshots as previously posted.

Dashboard -
Loads and previously created dashboard jguages still work and show readings / data, and update.
Can not create a new jgauge or edit an existing jgauge, the 'configure' code works but there are no feeds listed in the drop down box.

Schedule -
Displays standard 'no schedule created' text as I haven't used this part yet.  But still displays the page correctly.

Visualisations -
Page loads and, strangely enough, the feed drop down for works with all feeds displayed on all available vis elements.

App - My Electric -
App still functioning and updating

Log - 
[image]

MySQL -
The feed table has been deleted from mysql by emoncms.
Looking at the input table there are still references to feed 37 (as highlighted)

[image]

Once ',37:0' had been removed and the table saved, my inputs screen and feeds screen returned to what I had expected to see before the crash-

[image]

Having now deleted 14 & 15 and saved the input the system is back and working.

Hope this helps narrow down the issue.

Kind Regards
Andy Hodges

glynhudson, 2016-03-24:
Yes sorry I meant `sudo apt-get dist-upgrade`. It's a shame php5-redis is still at 2.2.5-1. Does anyone l know the ETA before it's updated to 2.2.7 which is already been tagged as 'stable'? 

@AndyRap  Have you updated php4-redis to 2.2.7 as detailed above?

AndyRap, 2016-03-24:
@glynhudson php4-redis or php5-redis.  If php4 then no, I haven't even checked for a version for php4.

I have had another crash.  Thought I had recovered, as per my notes 3hrs ago, but when I tried to add another feed it all went pear shaped again.

This time the visualisations link did not work.  The browser (EDGE) displayed a Hmmmmm.....  This page can not be displayed error with the usual retry and homepage links.

When viewing my inputs they were displayed as per my original screen shot post.  This time I clicked on the spanner to look at the feed logic and although it displayed, ALL 'log to feed' items showed a red error.

Upon deleting all feed logic from all the inputs I was then able to refresh the feeds screen and see a list of my feeds.  Upon deleting all the feeds I was left with two that would not delete.  In every field they just displayed 'unknown'.

I checked the feeds table and it was empty.

Could these be two virtual feeds that I had created?  Where does config data for the virtual feeds get stored, in mysql or in a file?  If a file then what is it called and where is it located?  If it's mysql then I am stumped because the feeds table was empty.  I even truncated it just in case that helped.

Regards
Andy Hodges

glynhudson, 2016-03-25:
Sorry I meant: have you updated **php5-redis** to 2.2.7? 

I believe virtual feeds are also stored in MYSQL, is the correct @chaveiro?

AndyRap, 2016-03-25:
Not on a working version yet Glyn.  Going to try that now.

I've just had it go again.  Here's what had been done this time.

Re-imaged SD card with same image as I have been using.  Booted Pi2.

Turn both of my power monitors on long enough for them to boot up and submit some data.  Then I turned them off.

Using the inputs now listed in EmonCMS I started to build my feeds and feed logic.  

As an example for feed 1 -
1. Log to feed -> Main Power - Real
2. - input -> Office Power - Real
2. Log to feed -> House Power - Real

and so on.

Are there any known issues with renaming a feeds 'tag' from the default "Node 1" to something more user friendly such as "Main" or "House" etc?

I have been doing this repeatedly with every install and just want to make sure I am not inducing an error.  The field is editable so I assumed it is simply part of the grouping of the structure.  What about when the structure does not have an equivalent inputs group?

Kind Regards
Andy Hodges

glynhudson, 2016-03-25:
No known error that I know about regarding reaming a feed. Please try and
update php5-redis, I experience lots of strange problems before updating.

AndyRap, 2016-03-25:
@glynhudson 

I have, as of 1am, been running the image again, this time with the updated redis.  At least I followed your instructions.  How can I check that the installed / running version is in fact the correct updated one?  Running the apt-get command just shows what's in the repository.

Just to be clear about the renaming of a feed.  I am looking at the renaming of the tag element rather than the name.

I will let you know how I get on :)

Kind Regards
Andy Hodges

AndyRap, 2016-03-25:
@glynhudson 

One thing I have noticed with this install and the updated redis.

When previously looking at the 'My Electric' app on either iPhone, iPad or Google Chrome, the charts, current power usage etc would automatically refresh every n seconds, this is not happening currently on any browser / platform.

Could this be related to the updated redis version?

Kind Regards
Andy Hodges

glynhudson, 2016-03-27:
If you don't mind I will close this issue since discussions are moving onto new issues. Please open a new issue or post on the OEM forums to continue discussion http://openenergymonitor.org/emon/forum/5

closed 2016-03-27 by glynhudson