emoncms/emoncms #497: Dashboard quick links not displaying on mobile portrait view

glynhudson, 2016-03-27:
I've just noticed when viewing Emoncms on my phone in portrait view I cannot see the quick links to my dashboards on the top RHS:

[image]

After rotating my phone to landscape view the dashboard links appear: 

[image]

Is this a design decision in Emoncms 9 to hide dashboard links when viewed on a small screen? Or a bug? It's not a big issue, but there is plenty of space on the top bar in portrait mode for at least a couple of the dashboard  star icons. How could I try and fix this?

Paul-Reed, 2016-03-27:
Glyn, you seem to be missing the dashboard icon.
It was a beta debated design decision to restrict the dashboard 'quick links' to bigger displays, to keep the workspace from looking cluttered, but you should still see the Dashboard icon to the left of the App icon as shown below. The dashboard icon dropdown should list all of your saved dashboards for easy access.
Have you tried deleting and re-installing the app module.

[image]

In landscape view, you should be able to see your quick links, but also still see the Dashboard icon.

[image]

elyobelyob, 2016-03-28:
I raised this a while back myself.
Can you explain the process of reinstalling the app module, and which
version of emoncms this applies to?
Am running core 9.3 myself. Thanks.

Paul-Reed, 2016-03-28:
To re-install the dashboard module;
`cd /var/www/emoncms/Modules`
`rm -rf dashboard`
`git clone https://github.com/emoncms/dashboard.git`
Because your dashboard configs are safely stored in MYSQL, reinstalling the dashboard module will not affect them - they won't be deleted!  
Paul

glynhudson, 2016-03-28:
Hi Paul,

This is interesting, I agree, with the design decision hiding the dashboard shortcuts on low screen widths is a good idea. However it seems on all my RaspberryPi / emonPi installs the Dashboard nav button is not visible. Re-installing the dashboard and app modules made no difference. On my local machine the Dashboard nav button IS visible! I've established the issue is nothing to do with the different settings.php files, testing the emonpi.settings.php file on my local machine made no difference to the Dashboard nav button. 

Can you think of anything else I can test to identify this issue? Do you have a RasPi install you could check?

Both my installs on on the same master branch and up-to-date

Paul-Reed, 2016-03-28:
I've emailed you a link to my site so that you can check if you wish.
Mine is also Master branch & up to date.

glynhudson, 2016-03-28:
Ah go it! It seems the Dashboard nav button disappears if all dashboards are published.

I've pushed a change which make the Dashboard button always visible if any dashboards exist even if they are all published. I've also renamed it to 'Dashboards' . 

Are you happy with this mode of operation? I think it's an improvement since at least mobile users will always be able to access their dashboards. 

https://github.com/emoncms/dashboard/commit/d62d0755d12ce223e50ac32b930d0f16b03cbcbe

closed 2016-03-28 by glynhudson

Paul-Reed, 2016-03-28:
Wasn't aware of that!
I wonder what the rationale was in including that check?

glynhudson, 2016-03-28:
I suppose if all dashboards are published therefore there are shortcut start icons for all the dashboards and the drop down menu is not required, which is fine until viewed on a small screen when everything is hidden!

elyobelyob, 2016-03-29:
That's a huge improvement on my phone, thanks. It would also be useful if
they retained the same order as the normal menus. In fact, I don't think
they can be ordered at the moment anyway? The first icon just clones, I was
hoping I could drag/drop those into a specific order.

Anyway, thanks for that small, but huge improvement.

elyobelyob, 2016-03-29:
Oh, I'd also remove the *'s in landscape mode on mobile. Without any naming
next to them, they were always a bit of guess work. This drop down does
away with the need for them.

I'd offer to help, but away at the moment. And front end stuff isn't really
my thing.

Thanks

Paul-Reed, 2016-03-29:
I'd keep the '*' in mobile landscape view, it's useful if you only have one quick link dashboard.

elyobelyob, 2016-03-29:
If you have one, there's no need for the drop down. Anywhere beyond three
and the drop down is required.
Also, can the drop down on desktop also be on the immediate left. Keep it
uniform.
Who's the guru on this stuff, is it you Paul? As mentioned, it's great. And
happy to see it so easily to update on my own system.

Paul-Reed, 2016-03-29:
Am I the guru?
I wish!
It was chaveiro who wrote v9
If you have one, there's no need for the drop down. Anywhere beyond three
and the drop down is required.
Also, can the drop down on desktop also be on the immediate left. Keep it
uniform.
Who's the guru on this stuff, is it you Paul? As mentioned, it's great. And
happy to see it so easily to update on my own system.