emoncms/emoncms #498: Json Error when upload bulk data

fohnbit, 2016-03-29:
Hello,

When I try to upload bulk data on my own Webspace (Hostet by HostEurope) I get:
Error: Format error, json string supplied is not valid

When I reduce the data to maximum 16, then it works.
If I upload 17 or more, I get the error above.

Using Version 9.31 | 2016.02.13

Upload data:
http://www.domain.com/emoncms/input/bulk.json?data=[[-2313,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2309,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2305,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2301,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2297,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2293,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2289,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2285,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2281,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2277,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2273,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2269,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2265,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2261,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2257,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2253,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2249,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2245,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2241,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2237,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2233,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2229,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2225,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2221,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2217,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2213,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2209,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2205,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2201,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2197,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2193,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2189,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2185,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2181,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2177,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2173,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2169,8,null,547.0,null,null,null,15.0,null,null,null,null]]&time=1459154173&apikey=xxx

fohnbit, 2016-03-29:
updated to 9.4 | 2016.03.15

same Problem.

On emoncms.org it works ... but not on my webspace ... where could be the problem?

Thank you!

fohnbit, 2016-03-29:
I added now:
       // Test Output
            $log = new EmonLogger(__FILE__);
            $log->info($_GET['data']);

to the file: "input_controller.php" at the beginning.

When I add more as 16 array´s ... I get an empty String :-(

Seems there is a problem with the http get upload.
Can I use also http Post and add the json in the post content?

Thank you!

chaveiro, 2016-03-29:
There is something limiting the maximal url character. Check your server configuration or use the POST method if possible.

fohnbit, 2016-03-29:
Ok, I try to send via POST.

On which URL I have to send?:
http://www.domain.com/emoncms/input/bulk.json
or
http://www.domain.com/emoncms/input/input_controller.php

Post Content are the json like:
data=[[-2313,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2309,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2305,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2301,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2297,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2293,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2289,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2285,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2281,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2277,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2273,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2269,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2265,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2261,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2257,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2253,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2249,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2245,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2241,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2237,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2233,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2229,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2225,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2221,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2217,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2213,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2209,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2205,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2201,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2197,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2193,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2189,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2185,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2181,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2177,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2173,8,null,547.0,null,null,null,15.0,null,null,null,null],[-2169,8,null,547.0,null,null,null,15.0,null,null,null,null]]&time=1459154173&apikey=xxx

Thank you!

fohnbit, 2016-03-29:
OK, I can now upload to:
/input/bulk.json
with POST

But I did not get a response back. Is this normal?
So I did not know if it uploaded succesfully.

With GET I receive an error Message or OK ... but with POST no reply.

My upload via POST:
POST /emoncms/input/bulk.json HTTP/1.1
Host: www.domain.com
User-Agent: JAVA 1.8
Connection: Keep-Alive
Accept-Encoding: gzip,deflate
Content-Type: application/x-www-form-urlencoded
Content-Length: 1319

data=[[-60,8,214.0,null,null,42.346,null,253.0,null,215.0,null,null],[-57,8,214.0,null,null,42.346,null,253.0,null,215.0,null,null],[-54,8,214.0,null,null,42.346,null,253.0,null,215.0,null,null],[-51,8,214.0,null,null,42.346,null,253.0,null,215.0,null,null],[-48,8,214.0,null,null,42.346,null,253.0,null,215.0,null,null],[-45,8,214.0,null,null,42.346,null,253.0,null,215.0,null,null],[-42,8,214.0,null,null,42.346,null,253.0,null,215.0,null,null],[-39,8,214.0,null,null,42.346,null,253.0,null,215.0,null,null],[-36,8,214.0,null,null,42.346,null,253.0,null,215.0,null,null],[-33,8,214.0,null,null,42.346,null,253.0,null,215.0,null,null],[-30,8,214.0,null,null,42.346,null,253.0,null,215.0,null,null],[-27,8,214.0,null,null,42.346,null,253.0,null,215.0,null,null],[-24,8,214.0,null,null,42.346,null,253.0,null,215.0,null,null],[-21,8,214.0,null,null,42.346,null,253.0,null,215.0,null,null],[-18,8,214.0,null,null,42.346,null,253.0,null,215.0,null,null],[-15,8,214.0,null,null,42.346,null,253.0,null,215.0,null,null],[-12,8,214.0,null,null,42.346,null,253.0,null,215.0,null,null],[-9,8,214.0,null,null,42.346,null,253.0,null,215.0,null,null],[-6,8,214.0,null,null,42.346,null,253.0,null,215.0,null,null],[-3,8,214.0,null,null,42.346,null,253.0,null,215.0,null,null]]&time=1459267187&apikey=xxx

chaveiro, 2016-10-01:
You should get the ok respnse anyway. Try to send an invalid json and see if an error is returned like with get.

closed 2017-02-28 by TrystanLea