emoncms/emoncms #503: Feeds disappeared after Feed operation definition....

arthur-esob, 2016-04-04:
Hello,
for an unknow reason and after an unknow combination of clicks, my feeds are gone, when I click on Feeds, I get the page saying "loading" and nothing happens.
My Emoncms is running locally on a CubieBoard. Everything worked well then suddenly.... nothing. It stopped working after defining feed operations (nothing fancy).
Can someone explain me how to clear all my feeds ? I don't have anything important so far so no real problem removing everything. I'd rather do this before removing Emoncms to reinstall it ...
Help would be really appreciated.
greetings from France
Pierre

chaveiro, 2016-04-06:
What do the logs say? Search for an error.

arthur-esob, 2016-04-06:
hello,
this is what the logger says when i click on "Feeds" 
2016-04-06 17:23:57.875|ERROR|process_model.php|__call() Call to undefined module method. Missing function on module? method=getstartday

any guess ? i must admnit i don't see any way of fixing this.
thanks for the help

Date: Wed, 6 Apr 2016 06:06:04 -0700

chaveiro, 2016-10-01:
Still have the issue ? Can you give instructions to reproduce?

arthur-esob, 2016-10-01:
Hello,
yes i still have the problem.
i completely removed emoncms then reinstal it. ever since i didn't have the same issue.
BUT i have a new one : i cannot focus correctly on the real time graph.
i must admit that those 2 problems drove me away from emoncms.
nevertheless, if you have any idea i'de be happy to test. there s no risk of loosing any data since i'm not using it anymore...
have a nice day

Date: Fri, 30 Sep 2016 19:14:45 -0700

chaveiro, 2016-11-10:
Nothing to fix.

closed 2016-11-10 by chaveiro