emoncms/emoncms #506: Problem with SSL

piertoni, 2016-04-18:
Is it still possible to publish data with simple http requests?
I have an electronic module that send data with http requests and tomorrow I have a presentation,
Today I had a cold shower trying to make the device work... it seems I can publish only with https and the device can't connect.
Is there a way to publish the data with old approach?
Thanks!

Paul-Reed, 2016-04-18:
This is more about how you have setup your SSL service than a emoncms issue. If you have blocked http you will only be able to publish data via https otherwise your server will block it.
If you have allowed both http & https access (which defeats the object of introducing https} then you can publish to either.

Paul

glynhudson, 2016-04-18:
Yes emoncms.org will still accept incoming posts from http. In fact most
incoming data us coming into emoncms.org as http as users have not updated
their devices

- sent from my mobile device

closed 2016-04-18 by glynhudson