emoncms/emoncms #511: No process description available for process 'Total pulse count to pulse increment'

ukmoose, 2016-04-27:
No process description available for process 'Total pulse count to pulse increment' with id '20'.
Add a description to Module\module_name\module_name_processlist.php in process_list() function, $list[] array at the 'desc' key.

jlm, 2016-06-29:
Bit me too.

glynhudson, 2016-06-29:
Good idea, any chance of a PR? I'm away on a break for the next couple of weeks with minimal Internet. If not, I'll take a look when I get back.

glynhudson, 2016-10-27:
I'm not sure what this input process does, any ideas @chaveiro @TrystanLea ? 

[image]

TrystanLea, 2017-02-28:
Added description https://github.com/emoncms/emoncms/commit/52ec995676da519353f08d4f6b77dd9926d9eb93

closed 2017-02-28 by TrystanLea