emoncms/emoncms #515: Detect version of emoncms from API

JumpMaster, 2016-05-03:
To keep track of changes to the API it should be possible to query the version of emoncms from the API.  This should either be returned with every request, have a dedicated request, or a known request such as feed list should return the version.

chaveiro, 2016-06-01:
A specific api for emoncms version info is recommended.
But i'd like APIS to not be versioned and changes to APIs ensure backward compatibility.
Meaning for example if you pass a new parameter to an old api that did't suported it at that time, the response just be empty or would not toke usage of that parameter.
What is your usecase?

JumpMaster, 2016-06-02:
Hi Chaverio,

I would like to use a newer API that was recently implemented but currently have no way of knowing if it is safe to use. If the Android app could get an emoncms version or API version I could use that to decide which API calls are compatiable.

Kev

chaveiro, 2016-06-02:
What newer API ?

JumpMaster, 2016-06-02:
For example /feed/data.json now has the mode option instead of using interval.  

e.g.
https://emoncms.org/feed/data.json?apikey=APIKEY&id=3&start=START&end=END&interval=86400&skipmissing=1&limitinterval=1

can be replaced with

https://emoncms.org/feed/data.json?apikey=APIKEY&id=3&start=START&end=END&mode=daily

TrystanLea, 2018-07-10:
This is now implemented https://github.com/emoncms/emoncms/pull/895

closed 2018-07-10 by TrystanLea