emoncms/emoncms #533: async: false on all ajax functions

XaroRSA, 2016-05-31:
Browser responses with : 
"Synchronous XMLHttpRequest on the main thread is deprecated because of its detrimental effects to the end user's experience".
Is there a reason why the async property is changed from default true to false?

chaveiro, 2016-05-31:
That's a warning.
The goal is to use async everywhere and slowly i had converted most of the widgets to it.
It helped a lot in responsiveness also.
There are places where it cant be done easy because of precedence concerns, or it just does not justify the effort.

XaroRSA, 2016-06-01:
Yes a warning indeed, thus the need to clarify why this warning is presented.
Do you have a example in the emoncms where this would cause a problem?

TrystanLea, 2016-06-01:
I dont think it causes any problems, its just not recommended best practice. It does however simplify the code somewhat to use synchronous requests.. which is why I often do it that way

closed 2016-06-15 by XaroRSA