emoncms/emoncms #540: Non-string variables returned as "strings" - part deux!

pb66, 2016-06-21:
#273 

When using the feed api to return the last value it is returned as a string eg "2222", I have been using a bash script to interact with emoncms and found the api call results are causing issues.eg

```
RESULT=$(curl -s "http://ukep.duckdns.org/feed/value.json?id=123&apikey=abc..123")
echo $((RESULT+2))
```
returns
```
-bash: "2222": syntax error: operand expected (error token is ""2222"")
```

Where as
```
RESULT="2222"
echo $((RESULT+2))
```
returns 2224 (as does setting RESULT=2222 without any quotes), the "included" quotes prevent the result being seen as numeric and then the quotes are accepted as part of the string resulting in double quotes. 

Since the api is calling the last VALUE shouldn't that be numeric (or null) not a string???

chaveiro, 2016-10-01:
Should be fixed, see https://github.com/emoncms/emoncms/pull/566/commits/4e2671eefe0c3d93d58ceea03f79597e50a79023

pb66, 2016-10-01:
No Change, not fixed, please re-open this issue!

Maybe due to the fix only being applied to MySQLTimeseries??

Querying PHPFina or PHPTimeseries or even mySQL results in a string containing double quotes to be returned as shown above, I've tested on at least one of every type of feed I have and none of them return actual numeric data.

chaveiro, 2016-10-01:
What api did you test? Please past the url.

Paul-Reed, 2016-10-01:
Isn't it posted in post#1 ?

pb66, 2016-10-01:
An example api would be "http://myemoncmsserver.com/feed/value.json?id=123&apikey=abc..123", I used it in the bash example to show how to experience the issue.

Using that api returns quote "123.45" unquote (when the feed value is 123.45) ie a string of 8 chars including the leading and trailing double quotes. It's therefore not considered numeric and the returned value cannot be used directly in math functions without removing the quotes and casting to a numeric datatype. Since the quotes are just useless baggage and provide no benefit, I see no reason for the value to be a quoted string.

@chaveiro please check #273, you have previously corrected other api's to return numeric data types rather than a string representation of a value.

@Paul-Reed, no correction needed, you got it.

Paul-Reed, 2016-10-01:
Having thought about this @pb66 ...
Isn't it because the same api returns non-numeric data too..
eg  "http://myemoncmsserver.com/feed/list.json&apikey=abc..123"

Paul

pb66, 2016-10-01:
/feed/list is a different endpoint to /feed/value, one returns a list and the other a value.

This was fixed before, again check #273.

Having re read that thread again, it was actually Trystan that [submited a fix](https://github.com/emoncms/emoncms/commit/afae89d00960a0e54a5d6dac5d71092d89694900) since then it has been changed a couple of times, more recently by @chaveiro https://github.com/emoncms/emoncms/commit/ac0407b0afc45a2760ea6eb1f778d3b4e56fda76#diff-a6d269aac8d41ae47febc430faf033ceL58 on the 2nd of June 16, a couple of weeks before I posted this issue, soon after I pulled in the changes.
There were obviously some changes in between these events as the method of casting to a float has altered, The original fix used "(float)" but at some point that changed to "1*" (multiply by 1 to cast to a number), this was removed again in June.

chaveiro, 2016-10-03:
Ok, so the problem is that php is not a strong typed language, meaning that it 'loses' track if the variable is numeric or string. The change trystan made by casting to float or *1 will fix the issue with numbers but not when the value is NULL as it will be converted to 0. Null is valid for some engines for processlist logic. 
Thet proposed change at https://github.com/emoncms/emoncms/issues/273 (with JSON_NUMERIC_CHECK) is tricky because it makes php guess by looking at variable content, i see one behaviour, the removing of leading zeros that could cause issues. But this could be tried out and see if there are complains.

On the other end. the 'conservative' fix should be made at the engine files level, instead of the controller and care taken if the engine can return a null. This fix is simple we need to hint php (by casting all know numeric variables to float) so that the json_encode function knows how to represent it.
The tricky part is that we need to do the cast is many places.
Also note note that it's not only the value but also other numeric fields that are  currently being returned as strings on other APIs (the time, the feed sizes, some ids).

chaveiro, 2016-10-03:
Lets see if this fixed anything on you side ?  https://github.com/emoncms/emoncms/pull/568
(i use mysql only, cant test other engines)

pb66, 2016-10-05:
Sorry I haven't had a chance to try this yet, but what you say makes sense.

If null values being cast to 0 is the issue would it be simpler to test for a null value (rather than test for a numeric value) and if false cast to numeric? that could possibly be done in just the one place before the api returns, although it is probably more correct to get the data in the correct datatypes upstream, it may leave holes for other data types to slip through the net, it not only needs implementing in multiple places but ongoing it needs maintaining in multiple places, plus any new engines or functions will need to comply too.

I hope to try this out as soon as i can, but I'm just cautious about doing "quick pulls" on a live server in case there are issues and no time to fix, i don't currently have a redundant "experimental" server instance, maybe at the weekend if not before.

glynhudson, 2016-10-06:
> I hope to try this out as soon as i can, but I'm just cautious about doing "quick pulls" on a live server in case there are issues and no time to fix, i don't currently have a redundant "experimental" server instance, maybe at the weekend if not before.

Have you tried the Emoncms docker instance? It can be fired up in minutes on any machine (assuming docker is installed) in minuites. It's great for testing as you can launch into a clean enviroment. I would love to get more people using it: https://github.com/emoncms/emoncms-docker

Another alternative (that I've started using recently) is cloud9 online IDE: https://community.openenergymonitor.org/t/emoncms-dev-in-a-browser-using-cloud9-ide/1756

*Off-topic: I woul really like to create a testing  enviroment as part of the emoncms-docker which automaticlly loads in a decent amount of test data and some complex dashboards at launch for testing and also a service to hit the API hard with a large volume of incoming data. However this is not the righ location to discuss this....*

pb66, 2016-10-09:
@nchaverio - Thanks again for looking at this, I have pulled in the latest changes and unfortunately there is still no change. I have tried 4 different feed types via the 2 different API calls

Realtime SQL feed using "feed/value" API (datatype is 1, engine is 0)
>"1476003315"

Realtime PHPfina feed using "feed/value" API (datatype is 1, engine is 5)
>"56.9375"

Realtime PHPtimeseries feed using "feed/value" API (datatype is 1, engine is 2)
>"56.9375"

Daily PHPtimeseries feed using "feed/value" API ("datatype is 2, engine is 2)
>"1.6662937868043"

All 4 of the above feeds in the same order using "feed/fetch" API 
> ["1476003315","56.9375","56.9375","1.6662937868043"]

(and just for my own ref the feed ids used for testing are 100231,100240,100236,100237)

Could we consider a single final test & cast in the API functions as I mentioned above? 
>"If null values being cast to 0 is the issue would it be simpler to test for a null value (rather than test for a numeric value) and if false cast to numeric?"

@glynhudson - I have answered you in "Using emoncms-docker. #7" and [on the forum](https://community.openenergymonitor.org/t/emoncms-dev-in-a-browser-using-cloud9-ide/1756/2?u=pb66)

chaveiro, 2016-10-14:
PB66 after the last update to 9.7.5 are you now getting the " removed?
I did test it here and they don't appear on feed/vale and feed/fetch and feed/data.

pb66, 2016-10-15:
I will try it later today or tomorrow, thanks @chaveiro.

pb66, 2016-10-15:
Hi @chaveiro - unfortunately not! the same issue persists (although #573 does appear fixed for me)

chaveiro, 2016-10-15:
That is very strange. What php version? Have low-write active?

pb66, 2016-10-15:
5.6.20-0+deb8u1 (Zend Version 2.6.0)
Full emoncms on a hdd, not low-write or read-only.

chaveiro, 2016-10-15:
I use 5.6.24 (Zend Version 2.6.0) so not a big difference.
Any others can confirm if the result are with or without "s ?

Examples:
/feed/fetch.json?ids=149,112
[1146.3259552456,350.43]

/feed/timevalue.json?id=149
{"time":1476555263,"value":1146.3225780929}

Paul-Reed, 2016-10-15:
feed/timevalue.json?id=13 returns;

{
time: "1476556170",
value: "233"
}

feed/fetch.json?ids=13,25 returns;

[
"148",
"243.28"
]

Raspberry Pi 2 running full emoncms in low-write - PHP Version => 5.6.24-0+deb8u1

pb66, 2016-10-15:
I had never tried the feed/timevalue api before so I just tried that and got
{"time":"1476556590","value":"31.745"}
I have even tried restarting the apache2 service o get a fresh start but that made no difference either.

chaveiro, 2016-10-15:
pb are you using redis?

pb66, 2016-10-15:
Yes I am using redid.

TrystanLea, 2017-02-28:
Looks like the difference was redis vs mysql method, should be fixed now in the latest commit:
https://github.com/emoncms/emoncms/commit/5b3d9db6a57a86cd3eedb8dff7a563073f584670

pb66, 2017-03-01:
Thanks @TrystanLea , When I get an opportunity to pull in the latest changes I'll test and let you know.

pb66, 2017-10-27:
This still isn't working.

https://someserver.com/feed/fetch.json?ids=100233,100234
returns
["23.51","-30.13779046944455"]

emoncms v9.8.8 | 2017.07.18
Linux 3.16.0-4-amd64 (Debian8 Jessie)
redis 2.8.17
php 5.6.30-0+deb8u1 (Zend Version 2.6.0)
MySQL 5.5.58-0+deb8u1
phpredis v3.1.4

glynhudson, 2017-10-27:
It's returning a numerical value for me. Tested on both local Emoncms and emoncms.org:

Here is the local emoncms result

`http://emonpi.local/emoncms/feed/fetch.json?ids=1,87`

Returns:

`[334,13]`

```
Emoncms 9.8.10 | 2017.08.17
Linux emonpi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux
redis 2.8.17
php 5.6.30-0+deb8u1 (Zend Version 2.6.0)
MySQL 5.5.57-0+deb8u1
phpredis v2.2.7
```

pb66, 2017-10-28:
That's odd! The most noticeable difference in specs (aside from platform and OS kernel) is phpredis versions, I wonder if that's got anything to do with it?
Just checked again in case I was going crazy and the feed/value api is also returning a string value the same as the  feed/fetch api

https://someserver.com/feed/value.json?id=100233
```
"268.02"
```
https://someserver.com/feed/fetch.json?ids=100233,100234,100236
```
["3208.45","5.492571484520123","3537.09"]
```

However it seems the feed/data api is still returning numeric values though

https://someserver.com/feed/data.json?id=100235&start=1509145615000&end=1509145715000&interval=10

```
[[1509145615000,46.430000305176],[1509145625000,45.615001678467],[1509145635000,44.720001220703],[1509145645000,46.169998168945],[1509145655000,44.915000915527],[1509145665000,43.865001678467],[1509145675000,43.119998931885],[1509145685000,42.610000610352],[1509145695000,43.450000762939],[1509145705000,44.540000915527],[1509145715000,44.165000915527],[1509145725000,42.459999084473]]
```

The feed/fetch api is essentially just multiple iterations of the feed/value api whilst the feed/data api is very different so that might help locate the issue.

I have several "balls in the air" right now and for a while I'll have no running test server I can experiment on so I may need to come back to this later.

JonMurphy, 2017-10-28:
another point for reference:
http://emonpi.local/emoncms/feed/fetch.json?ids=1,33 

```
["412","109"]   
```

(both phpfina)

http://emonpi.localdomain/emoncms/feed/data.json?id=1&start=1509207600000&end=1509207644000&interval=5
```
[[1509207600000,400],[1509207610000,385],[1509207620000,409],[1509207630000,406],[1509207640000,413],[1509207650000,390]]
```
&nbsp;

emonPi updated last on Wed Oct 18 16:13:28 CDT 2017
[image]

glynhudson, 2017-10-28:
Could you both try Emocnms 9.8.10 the current stable release version. There has been a number of changes to the API by @TrystanLea  and @chaveiro 

https://github.com/emoncms/emoncms/releases/tag/9.8.10

JonMurphy, 2017-10-28:
will do!

pb66, 2017-10-28:
That looks promising! 

Thanks @JonMurphy for confirming you saw it too (so I'm crazy after all) and for testing the latest version. I have a lot of unstashed changes that will make upgrading difficult at this point in time, if I stash the files now it will cause conflicts when I try and unstash. I will have to wait till I'm finished debugging and able to drop the changes and update emoncms.

It also confirms it's not a phpredis issue, cheers.

I will report back again when I update.

JonMurphy, 2017-10-28:
I think I confirmed you are NOT crazy.  But I am not a Doctor.  And we've never met.  And I don't know you.  So I'm probably wrong.  Who's to say...   
;-)

Paul-Reed, 2018-02-11:
No updates since October so closed.
If there are any outstanding issues, please let us know.

closed 2018-02-11 by Paul-Reed